verilog-to-routing/vtr-verilog-to-routing #531: Improve VPR Regression Test Coverage

kmurray, 2019-04-08:
There are a varitey of VPR related options that remain untested. It would be good to come up with fast regression tests which can go into vtr_reg_strong to cover some of these:

- [--verify_binary_search](https://docs.verilogtorouting.org/en/latest/vpr/command_line_usage/#cmdoption-vpr-verify-binary-search)
- Gaussian & Delta channel width variants](https://docs.verilogtorouting.org/en/latest/arch/reference/#tag-global-routing-information-x)
- [All supported SDC commands](https://docs.verilogtorouting.org/en/latest/vpr/sdc_commands/)
- [Reading pad locations from a file](https://docs.verilogtorouting.org/en/latest/vpr/command_line_usage/#cmdoption-vpr-fix-pins)
- [Full stats](https://docs.verilogtorouting.org/en/latest/vpr/command_line_usage/#cmdoption-vpr-full-stats)

This list is not extensive. There are likely many other areas which could be better tested.

Regression test code coverage is a good way to identify these.

vaughnbetz, 2019-04-18:
Good summer student task ... lets reassign when they start.

kmurray, 2019-05-31:
@vaughnbetz @renrichard @tinayang3024 @suyang5 Anyone interested in looking into adding these tests?

vaughnbetz, 2025-04-29:
This is a good first summer student task.