verilog-to-routing/vtr-verilog-to-routing #533: Not able to make in vtr_release

shubhamrai26, 2019-04-09:
Make is ending up with errors. 
[compile_error.log](https://github.com/verilog-to-routing/vtr-verilog-to-routing/files/3057925/compile_error.log)

#### Expected Behaviour

It should have shown 100% complete

#### Current Behaviour

#### Possible Solution

#### Steps to Reproduce

1. 
2. 

#### Context

#### Your Environment

* VTR revision used: VTR 7
* Operating System and version: Linux MINT 19 Cinnamon, kernel - 4.15.0-46-generic
* Compiler version: gcc7.3.0

kmurray, 2019-04-09:
Your log file doesn't contain any errors (perhaps you only captures stdout and not stderr?) so I can't see what the compilation issue is.

However, the VTR 7 release is rather old and was never tested on more modern versions of Linux. My suggestion would be to try the latest trunk version of VTR. You can see more information on building VTR [here](https://docs.verilogtorouting.org/en/latest/dev/building/).

shubhamrai26, 2019-04-09:
Exactly that was my point. I downloaded the VTR release from the link mentioned and just did a make. There was no error which was shown but it just failed at the ODIN. I then downloaded the latest from the git repository and atleast I am able to compile and do stuff. I would request you to please upload a stable release of VTR in the download link--  http://www.eecg.utoronto.ca/vtr/terms.html

vaughnbetz, 2019-04-18:
We're working on a VTR 8.0 release -- probably a couple of months away though.

jeanlego, 2020-03-31:
for future reference, VTR 7.0 does compile successfully with GCC 4.9, anything above seems to fail.
@kmurray I think this can be closed?

kmurray, 2020-03-31:
Yep @jeanlego now that VTR 8 is [released](https://verilogtorouting.org/download), we can close this.

closed 2020-03-31 by kmurray