verilog-to-routing/vtr-verilog-to-routing #537: check_route appears to be fairly slow

litghost, 2019-04-19:
#### Expected Behaviour

Given that ``check_route`` should just be a sanity check, I would expect it to be a fast check.  

#### Current Behaviour

Currently ``check_route`` appears to take a long as the routing step!

Runtime of routing with calls ``check_route`` commented out:
```
The entire flow of VPR took 374.07 seconds (max_rss 5237.5 MiB)
```
Runtime of routing with ``check_route`` not commented out:
```
The entire flow of VPR took 635.03 seconds (max_rss 5162.6 MiB)
```

That's approximately 261 dedicated to ``check_route`` that seems unreasonably high.

#### Possible Solution

1. Provide command line option to disable ``check_route``
2. Profile and optimize ``check_route``.  I suspect there is a O(n^2) or O(n) with large time constant algorithm lurking in ``check_route``.

#### Steps to Reproduce

1. Example from https://github.com/verilog-to-routing/vtr-verilog-to-routing/issues/526 demostrates this issue.  Run with VPR with ``check_route`` commented out and run again without ``check_route`` commented out.
```
vpr \
 arch.unique_pack.xml \
 top.eblif \
 --device xc7a50t-basys3-test --read_rr_graph rr_graph_xc7a50t-basys3_test.rr_graph.real.xml \
 --min_route_chan_width_hint 100 --max_criticality 0.0 --max_router_iterations 500 --routing_failure_predictor off --router_high_fanout_threshold -1 \
 --constant_net_method route --route_chan_width 500 --clock_modeling route --place_algorithm bounding_box --enable_timing_computations off \
 --allow_unrelated_clustering on --route
```

With ``check_route``: ``The entire flow of VPR took 147.16 seconds (max_rss 6301.9 MiB)``
Without ``check_route``: ``The entire flow of VPR took 74.74 seconds (max_rss 6302.2 MiB)``

#### Context

This issue doubly affects us because ``check_route`` in master is currently called 3 times!!!  See https://github.com/verilog-to-routing/vtr-verilog-to-routing/pull/523 to remove the duplicate call.  With https://github.com/verilog-to-routing/vtr-verilog-to-routing/pull/523, ``check_route`` is called twice in our flow, once during routing, and once when the routing is read when invoking ``genfasm``.  If ``check_route`` is fast, this double invocation isn't a problem, but it is very slow.

#### Your Environment

* VTR revision used:
* Operating System and version:
* Compiler version:

kmurray, 2019-04-22:
Thanks for the report.

I also wouldn't expect `check_route` to be _that_ slow. I'd expect it to be `O(n)` where `n` is the size of the RR graph.

Now on large RR graphs this will take time, but I'd expect it to be much less than the actual route times (which is not what you are seeing).

kmurray, 2019-04-23:
Thinking about this some more, I believe your architectures make fairly extensive use of non-configurable edges.

`check_route` includes relatively 'new' checks to verify the correctness of routing using non-configurable edges. However they've only been tested on architectures with relatively few non-configurable edges -- so that would be one of the first things to look into.

closed 2025-05-15 by github-actions[bot]