verilog-to-routing/vtr-verilog-to-routing #538: Import SymbiFlow/fasm into libs/EXTERNAL

mithro, 2019-04-19:
#### Proposed Behaviour

Using git-subtrees we should import the [SymbiFlow/fasm](https://https://github.com/SymbiFlow/fasm) repository to [libs/EXTERNAL/fasm](./libs/EXTERNAL).

This would allow us to;
 * use the fasm tools in regression testing to check that `genfasm` utility is producing valid output
 * import the fasm file specification into the vpr file formats section of the documentation
 * make sure that SymbiFlow and VtR stay in sync around the FASM file format
 * provide a place to put anything FASM specific that can be used with anything FASM related.
 * demonstrate some concrete collaboration between VtR and SymbiFlow community
 * other things I haven't thought of....

#### Context

FASM support was landed a while ago. This would is an extension of that.

kmurray, 2019-05-06:
One disadvantage of this approach is that since genfasm depends on libvpr any changes to the libvpr API (which has no stability guarantees) would break the build until genfasm was updated. 
(Note that the genfasm dependency is in the opposite direction from the other libraries in libs/EXTERNAL, upon which vpr etc depend -- so this is a different case).
As a result, my tendency is to keep the genfasm code in the VTR tree directly so it can be refactored directly when libvpr changes.

Of course it makes sense that symbiflow contributors should be able to update/modify genfasm directly in the VTR source tree (since you folks are really the owners of that code).

Effectively I think we should have a system for controlling commit privileges at a finer granularity than the full repo. Contributors could propose PRs outside of the directories they have ownership of, but those would need to be reviewed and approved by the relevant owners of those directories.

I think this should be possible with Github's [protected branches](https://help.github.com/en/articles/configuring-protected-branches), [required pull request reviews](https://help.github.com/en/articles/enabling-required-reviews-for-pull-requests) and [code owners](https://help.github.com/en/articles/about-code-owners). However I haven't had the time to figure out how to make that work.

@mithro Any thoughts?

@vaughnbetz FYI

mithro, 2019-05-07:
@kmurray 

I'm not suggesting we move `genfasm` under `libs/EXTERNAL/fasm`. That should remain in the current location and part of the main vpr code.

I'm suggesting that we import fasm repository which includes *extra* tooling and documentation. That repo will eventually include tools for checking that produced fasm files are valid, style guides, etc. I expect we want to use this stuff in the testing scripts.

closed 2025-05-15 by github-actions[bot]