verilog-to-routing/vtr-verilog-to-routing #543: generate_cmos_tech_data.pl giving the following error "Could not find fallt in spice output"  with 10nm PTM

shubhamrai26, 2019-04-23:
I m trying to generate an xml file for 10 nm PTM CMOS model. I m getting the following error.
"

 Could not find fallt in spice output"
Is it important ? or is there any workaround ?  
#### Expected Behaviour

#### Current Behaviour

#### Possible Solution

#### Steps to Reproduce

1. i have uploaded a file in text format. 
2. just change the extension to .pm to make it work. 

#### Context

#### Your Environment

* VTR revision used:
* Operating System and version:
* Compiler version:

shubhamrai26, 2019-04-23:
[ptm_hp_10nm.txt](https://github.com/verilog-to-routing/vtr-verilog-to-routing/files/3107728/ptm_hp_10nm.txt)

kmurray, 2019-04-23:
Thanks for the report.

@jgoeders can you take a look?

shubhamrai26, 2019-04-25:
Is it possible to circumvent this issue by hack or something ?  I am okay with some degradation in results.

closed 2025-05-15 by github-actions[bot]