verilog-to-routing/vtr-verilog-to-routing #544: Allow placement macros to be swapped

kmurray, 2019-04-24:
#### Current Behavior

Currently, if a placement macro tries to swap with another placement macro the swap is immediately aborted: https://github.com/verilog-to-routing/vtr-verilog-to-routing/blob/3c5c6f921db6f927056d11aa00910e40b0d882bf/vpr/src/place/place.cpp#L1149-L1154

#### Proposed Behavior

The code should be extended to allow the swapping of two macros.

#### Context

Forbidding two macros to be swapped likely hurts quality, particularly since it prevents common macros (like carry-chains) from being swapped. This would be come particularly important in designs with a high ratio of carry-chains to regular logic as many swaps would be aborted.

kmurray, 2019-04-24:
@vaughnbetz FYI

kmurray, 2019-05-01:
As of 400434bb6a4f5810f9b25583189897ee4bd8a3ac macro swapping is now supported.

We currently support: macro-macro swaps and macro self swaps (i.e macro shifts) between pairs of macros. Currently any macros being swapped to must be 'contained' in the macro being swapped from.

closed 2019-05-01 by kmurray

vaughnbetz, 2019-05-01:
Awesome!  Thanks.