verilog-to-routing/vtr-verilog-to-routing #550: Fanout delay modelling

litghost, 2019-05-01:
#### Proposed Behavior

Add a new buffered configurable switch type that's input capacitance is only added to upstream nodes if the switch is enable.

#### Current Behavior

Currently the buffered configurable switch input capacitance is modeled as adding the max Cin to the node that drives the switches.  This results in a static fanout-capacitance  model, where the loading capacitance is constant no matter how many output switches are used (or unused).

#### Possible Solution

Add a new switch type, and during capacitance propagation (e.g. https://github.com/verilog-to-routing/vtr-verilog-to-routing/blob/master/vpr/src/timing/net_delay.cpp#L347), add the switch input capacitance if the switch is in use.

It is worth noting that this model can be implemented using pass-through switches in front of buffered switches, but that will double the number of edges, so that seems like a poor solution.

#### Context

The 7-series routing timing model appears to increase the capacitance because on the number of active output switch.  Adding this switch type and modifying the downstream capacitance calculation would appear to be the easiest way to approximate the timing model.

vaughnbetz, 2019-05-01:
I'd probably implement this as a new "internal" capacitance that is added only when the switch is enabled.  You are right that there is some internal capacitance in the pass transistor muxes that is exposed when the switch is on (for direct-drive routing, which V7 uses).  

Could be called Cinternal, or Cint_enabled or some such and added to the arch file.  And then needs to be added into the route_tree_timing timing modeling. This is doable, and I'd prefer to add it as an entirely new type of capacitance to avoid overloading Cin (which still exists in direct-drive routing, and is the input capacitance you can't isolate).

How large is the impact?  This is a noticable but not overly large (usually ~5% - 10% delay variability due to loading) effect in Altera chips.  So you can probably ignore it in the first iteration of your timing model without it being too bad.

vaughnbetz, 2019-05-01:
Good task for a summer student, and we have 3 starting next week.

litghost, 2019-05-01:
> I'd probably implement this as a new "internal" capacitance that is added only when the switch is enabled. You are right that there is some internal capacitance in the pass transistor muxes that is [...]

Assuming our approximation of the V7 timing model is correct, there is no Cin at all!  All of the loading is either intrinsic or Cinternal.  Our approximation of the model is not 100% accurate.

As for the effect of the Cinternal, it varies based on fanout.  In low fanout cases, you are right that the added delay is small.  However on large fanout nets, the routing delay can increase by 20-30%.

litghost, 2019-05-01:
As a concrete example from the model

Net intrisic delay (no capacitive loading): 983 ns
Net delay (including capacitve loading): 1165 ns

So in this example, ~15% increase in delay.

vaughnbetz, 2019-05-01:
Those numbers look about right (high fanout nets can get up into the +20 - 30% range in Altera chips too).

You are right that you never really need Cin.  It's a convenience for adding up switch capacitance attached to wires when you specify an architecture-- you can equivalently just increase the wire capacitance, or the switch delay, to account for the loading on a wire due to switches hanging off it.  Basically you have more degrees of freedom than you need when specifying a timing model, so you can choose to make all the switch Cin values 0 if you wish and still fit the delay well.

litghost, 2019-05-01:
> Those numbers look about right (high fanout nets can get up into the +20 - 30% range in Altera chips too).
>
> You are right that you never really need Cin. It's a convenience for adding up switch [...]

This is a good sanity check.  Cinternal shouldn't be too hard to add either.

If you have a student to take the task, this is probably a good isolated task for them.  My team can work on our general model import code, and once VPR has Cinternal support, the model quality will improve once the tag is understood.  Thanks!

vaughnbetz, 2019-05-01:
Richard, I think this is a good initial task in VTR.  Lets discuss next week.

renrichard, 2019-05-02:
Thank you, looking forward to discuss the task next week.

kmurray, 2019-07-31:
I believe this is now supported with #607, so I'm going to close the issue.

@litghost Please give the Cinternal support a try to see if it meets your needs.

closed 2019-07-31 by kmurray