verilog-to-routing/vtr-verilog-to-routing #551: Process corner modelling

litghost, 2019-05-01:
#### Proposed Behaviour

Timing models are generally preformed at a particular process corner (and temperature).  VTR currently only supports one process corner.  Using a slow process corner is likely sufficient if slack timing is the dominate issue during routing, but in fast corners, I believe hold timing violations become possible.

In addition to process corner, temperature affects may be significant if the design is operating over a large temperature range.

Enabling multiple versions of timing models (at a minimum a fast and slow model) would enable corner analysis and optimization on designs that require that fidelity.

#### Current Behaviour

A single process corner is support.

#### Possible Solution

#### Context

7-series timing models have both SLOW and FAST corners.  Other parts may have cold and fast timing models as well.  Analysis is possible on a routed design, but PnR optimization can at most be run at one corner.

litghost, 2019-05-01:
This is likely a low priority change, but something to think about putting on the road map.

kmurray, 2019-05-02:
Agreed, this is a good thing to put on the roadmap.

One item I know has been discussed is better encapsulation of the device delay model. At the moment the delay model is spread through-out the code base, some in the PB type hierarchy and some in the RR graph. To support this it probably makes sense to factor the delay model out of those components and into a separate encapsulated data structure (which can be easily swapped for a different PVT, or multiple PVTs supported).

kmurray, 2019-05-02:
Another note, is that Tatum (VTR's STA library) can be extended to perform simultaneous PVT analysis efficiently (via SIMD). That code never made it back into master, but is something to re-consider at the point where fast multi-PVT analysis is required (i.e. during optimization).

lnsharma, 2021-02-23:
From eFPGA fabric perspective, it is important to do STA on different process corners. I am wondering if this issue is still in lower priority or are you planning to address this in near future? We are currently using VTR for our new QuickLogic eFPGA devices and this feature is really helpful for our users.