verilog-to-routing/vtr-verilog-to-routing #552: Classic VPR and Tatum critical paths do not match

shubhamrai26, 2019-05-02:
I am trying VTR for EPFL benchmarks with power arch xml file
#### Expected Behaviour

Every benchmark in EPFL benchmark suite should pass. 
#### Current Behaviour

Two benchmarks hyp and sqrt showed the following error in **vpr_stdout.log** "Classic VPR and Tatum critical paths do not match". Other benchmarks work flawlessly. 
#### Possible Solution

#### Steps to Reproduce

 k6_N10_I40_Fi6_L4_frac0_ff1_22nm.xml
1. Adding Vpr_stdout log for both. 
2. 

#### Context

#### Your Environment

* VTR revision used: VTR 8
* Operating System and version: Linux Mint 19 with 4.15 kernel
* Compiler version: 
[vpr_stdout.log](https://github.com/verilog-to-routing/vtr-verilog-to-routing/files/3138801/vpr_stdout.log)
[vpr_stdout.log](https://github.com/verilog-to-routing/vtr-verilog-to-routing/files/3138803/vpr_stdout.log)

kmurray, 2019-05-02:
Thanks for the report.

The differences are very small and I expect are just the result of floating-point round-off differences, so I wouldn't worry about them.

My suggestion would be to disable the classic timing analyzer (it is no longer supported). You can do that by commenting out `#define VPR_ENABLE_CLASSIC_STA` in vpr_types.h.

I'll also point out that the classic timing analyzer has been removed in the latest revisions of VTR.

shubhamrai26, 2019-05-02:
Thanks for the quick response. After commenting out #define VPR_ENABLE_CLASSIC_STA in vpr_types.h, do i need to rebuild VTR ?

kmurray, 2019-05-03:
Yep, after you rebuild there should be no more calls to the classic STA engine and the errors you were encountering should go away.

closed 2019-05-28 by kmurray