verilog-to-routing/vtr-verilog-to-routing #556: VPR fails a assertion in find_most_common_block_type when only using `.subckt`

mithro, 2019-05-06:
When using an architecture which has only `.subckt` VPR fails an assert in `vpr/src/util/vpr_utils.cpp`

Output is included below;
```
RouterOpts.max_criticality: 0.990000
RouterOpts.routing_failure_predictor = SAFE
RouterOpts.routing_budgets_algorithm = DISABLE

AnalysisOpts.gen_post_synthesis_netlist: false

RoutingArch.directionality: BI_DIRECTIONAL
RoutingArch.switch_block_type: WILTON
RoutingArch.Fs: 3

# Packing
Begin packing '/github/SymbiFlow/symbiflow-arch-defs/build/utils/vlog/tests/fig41-full-adder/adder.test.eblif'.

After removing unused inputs...
        total blocks: 6, total nets: 5, total inputs: 3, total outputs: 2
Begin prepacking.
Finish prepacking.
Using inter-cluster delay: 1.2887e-09
/github/verilog-to-routing/vtr-verilog-to-routing/vpr/src/util/vpr_utils.cpp:776 find_most_common_block_type: Assertion 'max_type != nullptr' failed.
```

Architecture XML file;
```xml
<?xml version="1.0"?>
<!-- set: ai sw=1 ts=1 sta et -->
<architecture xmlns:xi="http://www.w3.org/2001/XInclude">
  <models>
    <model name="adder">
      <input_ports>
        <port combinational_sink_ports="cout sum" name="a"/>
        <port combinational_sink_ports="cout sum" name="b"/>
        <port combinational_sink_ports="cout sum" name="cin"/>
      </input_ports>
      <output_ports>
        <port name="cout"/>
        <port name="sum"/>
      </output_ports>
    </model>
  </models>
  <complexblocklist>
    <pb_type name="IBUF">
      <output name="I" num_pins="1"/>
      <pb_type blif_model=".input" name="PIN" num_pb="1">
        <output name="inpad" num_pins="1"/>
      </pb_type>
      <interconnect>
        <direct input="PIN.inpad" name="P" output="IBUF.I"/>
      </interconnect>
      <pinlocations pattern="custom">
        <loc side="right" xoffset="0" yoffset="0">IBUF.I</loc>
      </pinlocations>
    </pb_type>
    <pb_type name="OBUF">
      <input name="O" num_pins="1"/>
      <pb_type blif_model=".output" name="PIN" num_pb="1">
        <input name="outpad" num_pins="1"/>
      </pb_type>
      <interconnect>
        <direct input="OBUF.O" name="P" output="PIN.outpad"/>
      </interconnect>
      <pinlocations pattern="custom">
        <loc side="left" xoffset="0" yoffset="0">OBUF.O</loc>
      </pinlocations>
    </pb_type>
    <!-- pb_type.xml file will be included here -->
    <pb_type height="1" name="TILE" width="1">
      <pb_type blif_model=".subckt adder" name="adder" num_pb="1">
        <input name="a" num_pins="1"/>
        <input name="b" num_pins="1"/>
        <input name="cin" num_pins="1"/>
        <output name="cout" num_pins="1"/>
        <output name="sum" num_pins="1"/>
        <delay_constant in_port="adder.a" max="300e-12" out_port="adder.cout"/>
        <delay_constant in_port="adder.b" max="300e-12" out_port="adder.cout"/>
        <delay_constant in_port="adder.cin" max="10e-12" out_port="adder.cout"/>
        <delay_constant in_port="adder.a" max="300e-12" out_port="adder.sum"/>
        <delay_constant in_port="adder.b" max="300e-12" out_port="adder.sum"/>
        <delay_constant in_port="adder.cin" max="300e-12" out_port="adder.sum"/>
      </pb_type>
      <pinlocations pattern="custom">
        <loc side="left" xoffset="0" yoffset="0">TILE.a TILE.b TILE.cin </loc>
        <loc side="right" xoffset="0" yoffset="0">TILE.cout TILE.sum </loc>
      </pinlocations>
      <interconnect>
        <direct input="TILE.a" name="TILE.a-adder.a" output="adder.a"/>
        <direct input="TILE.b" name="TILE.b-adder.b" output="adder.b"/>
        <direct input="TILE.cin" name="TILE.cin-adder.cin" output="adder.cin"/>
        <direct input="adder.cout" name="TILE.cout-adder.cout" output="TILE.cout"/>
        <direct input="adder.sum" name="TILE.sum-adder.sum" output="TILE.sum"/>
      </interconnect>
      <input equivalent="none" name="a" num_pins="1"/>
      <input equivalent="none" name="b" num_pins="1"/>
      <input equivalent="none" name="cin" num_pins="1"/>
      <output equivalent="none" name="cout" num_pins="1"/>
      <output equivalent="none" name="sum" num_pins="1"/>
    </pb_type>
  </complexblocklist>
  <layout>
    <!-- layout will be auto generated here -->
    <fixed_layout height="5" name="device" width="7">
      <!--
   0123456
 0 .......
 1 .I.T.O.
 2 .I.T.O.
 3 .I.T...
 4 .......
-->
      <single priority="1" type="IBUF" x="1" y="1"/>
      <single priority="1" type="IBUF" x="1" y="2"/>
      <single priority="1" type="IBUF" x="1" y="3"/>
      <single priority="1" type="TILE" x="3" y="1"/>
      <single priority="1" type="TILE" x="3" y="2"/>
      <single priority="1" type="OBUF" x="5" y="1"/>
      <single priority="1" type="OBUF" x="5" y="2"/>
    </fixed_layout>
  </layout>
  <device>
    <sizing R_minW_nmos="6065.520020" R_minW_pmos="18138.500000"/>
    <area grid_logic_tile_area="14813.392"/>
    <connection_block input_switch_name="mux"/>
    <switch_block fs="3" type="wilton"/>
    <chan_width_distr>
      <x distr="uniform" peak="1.000000"/>
      <y distr="uniform" peak="1.000000"/>
    </chan_width_distr>
    <default_fc in_type="frac" in_val="1.0" out_type="frac" out_val="1.0"/>
  </device>
  <switchlist>
    <switch Cin=".77e-15" Cout="4e-15" R="551" Tdel="58e-12" buf_size="27.645901" mux_trans_size="2.630740" name="mux" type="mux"/>
  </switchlist>
  <segmentlist>
    <segment Cmetal="22.5e-15" Rmetal="101" freq="1.000000" length="1" name="local" type="bidir">
      <sb type="pattern">1 1</sb>
      <cb type="pattern">1</cb>
      <wire_switch name="mux"/>
      <opin_switch name="mux"/>
    </segment>
  </segmentlist>
</architecture>
```

`.eblif file`
```blif
model top
.inputs a b cin
.outputs cout sum
.subckt adder a=a b=b cin=cin cout=cout sum=sum
.cname adder
.end
```

#### Expected Behaviour

Things work.

#### Current Behaviour

Assertion happens and things fail.

#### Possible Solution

The issue seems to be that `grid.num_instances` is empty;
```
name:num_instances type:0x55df698042f8 capacity:0
name:num_instances type:0x55df69804400 capacity:0
name:num_instances type:0x55df69804508 capacity:0
name:num_instances type:0x55df69804610 capacity:0
```

Earlier grids have the following;
```
name:devicenum_instances type:0x55df698042f8 capacity:0
name:devicenum_instances type:0x55df69804400 capacity:3
name:devicenum_instances type:0x55df69804508 capacity:2
name:devicenum_instances type:0x55df69804610 capacity:2
```

mithro, 2019-05-06:
The issue seems to be this code;
https://github.com/verilog-to-routing/vtr-verilog-to-routing/blob/fe33cd05f4c9aa03fe457926270dca1c336bf5eb/vpr/src/pack/pack.cpp#L297-L330

mithro, 2019-05-06:
Current work around seems to be to add `      --target_ext_pin_util 1.0,1.0`

closed 2019-05-06 by kmurray

mithro, 2019-05-07:
@kmurray Any idea why it doesn't figure out that `TILE` is the most common tile?

kmurray, 2019-05-07:
When we try to infer the logic block type, we usually find the block type which contains .names or .latch, which don't exist on this architecture.

At that point the code falls back to looking for the most common block type in the device grid.
However since this is actually called *before* packing proper starts the device grid hasn't actually been constructed yet.

A better fix would be to construct a dummy grid (e.g. 10x10) and use that to look for the common grid type. We currently do that for [estimating the inter-cluster delay](https://github.com/verilog-to-routing/vtr-verilog-to-routing/blob/master/vpr/src/base/vpr_api.cpp#L848-L849( before packing so some thing similar could be done here if the device grid was found to be empty . However I haven't had time to work that up, if you want to tackle it feel free to make a PR to that effect.

mithro, 2019-05-07:
We should log an issue so someone knows they need to do what you said. Unsure if I'll have time too...

commit 639bf699 (2019-05-06): vpr: If logic block type can't be identified default to 100% pin util targets

Previously would halt with an assertion error. We now produce warning
messages instead (warning that this may produce a denser packing than
desired) and allow the tool to continue using the relaxed targets.

Closes #556