verilog-to-routing/vtr-verilog-to-routing #563: Placement delta delay block choice seems limited

litghost, 2019-05-13:
#### Expected Behavior

Placement timing should sample a variety (all?) of block types when forming delta matrix,.

#### Current Behavior

Delta delays are sample at most 5 types of blocks, and the actually variation is limited by the geometry of the grid.

Different blocks samples:
- https://github.com/verilog-to-routing/vtr-verilog-to-routing/blob/master/vpr/src/place/timing_place_lookup.cpp#L394
- https://github.com/verilog-to-routing/vtr-verilog-to-routing/blob/master/vpr/src/place/timing_place_lookup.cpp#L423
- Blocks at (x_low, y_high), (x_low, y_low) etc 
https://github.com/verilog-to-routing/vtr-verilog-to-routing/blob/master/vpr/src/place/timing_place_lookup.cpp#L455

#### Context

The delta matrix in my architecture samples two locations:
 - An output port (as a source)
 - One SLICEL

As a result of such a limited sampling of the grid, many locations are have poor delta delays.  Just imagine if the blocks chosen were special (think global clock drive blocks or constant sources, e.g. GND).  The delay matrix would be very skewed two that part of the interconnected.

#### Possible Solution

I'm not confident in this solution, but it seems to me that the delay matrix should sample each **type** of block, e.g. LUT, RAM, IO, CLK, DSP, etc.  This will force variaty into the sample delay values over the various types of interconnect and avoid picking "bad" examples.

Alternatively a delay matrix per source type might be suitable, because it will then better describe the delays for that specific block type.  This approach falls down if OPIN's have special routing.  Think PLL outputs, some of which could go to dedicated clock networks, and some might be just status pins (like LOCK).

litghost, 2019-05-13:
It is worth noting that this problem was identified trying to diagnose the following error:
```
vtr-verilog-to-routing/libs/EXTERNAL/libtatum/libtatum/tatum/timing_paths.cpp:45 find_critical_paths: Assertion 'slack.valid()' failed.
```

It is unclear if a poor sampling of the delay matrix is the actual problem, but it seems reasonable to think if the delay matrix is corrupt as a result of poor sampling, that it might cause further issues.

litghost, 2019-05-13:
I've isolated the issue to a couple of factors:
 - The current delay matrix sampling methodology seems poor.  A more comprehensive sampling seems warrented.  If the amount of time that takes is too large, maybe the delay matrix can be computed once per rr graph and saved between placement tasks.
 - The architecture that is being examined has "holes" in the grid.  As an approximation for discussion, the columns are roughly:

```
LUT SLICE | LUT SLICE  | EMPTY | EMPTY | LUT SLICE | LUT SLICE repeat
```

Those two empty columns can be removed, it seems like the delay matrix should tolerate non-uniform grids, and it currently cannot.  As a preliminary suggestion, does it make sense to use a nearest neighboor approach to fill "holes" in the delay matrix?  On my side, I can work to remove those "EMPTY" columns through manipulation of the grid.

litghost, 2019-05-14:
I just ran into a case where the sampling algorithm had a problem because of an asymmetry in the grid.  An input pad was located at the lower right corner (e.g. high x and high y), but nothing else was in the grid near it.  Because of how sampling worked, the delay matrix didn't include this "far" point in the sampling.

Idea for a better delay matrix sampling: Sample 1 block of each type at min(x) / min(y) and max(x) / max(y)

This ensures delay matrix coverage from both a spatial standpoint and a routing variety standpoint.

kmurray, 2019-05-14:
Undoubtedly a better job could be done at sampling the delay map, particularly for highly non-uniform grids with empty holes, special limited connectivity blocks etc. At the moment that hasn't been a significant issue on the architectures we usually work with with so no one has written the code for that, feel free to give it a try :)

The current code tries to initially [sample from the edges of the chip](https://github.com/verilog-to-routing/vtr-verilog-to-routing/blob/master/vpr/src/place/timing_place_lookup.cpp#L394-L447) to ensure it captures the longest delta distances (e.g. across the chip) which can't be generated by sampling from a more uniform central portion of the chip. It then [re-samples from the mid-points](https://github.com/verilog-to-routing/vtr-verilog-to-routing/blob/master/vpr/src/place/timing_place_lookup.cpp#L450-L496) to avoid edge effects (e.g. partial length wires that are clipped by the edge of the device), and sample from more common block types (e.g. fewer odd blocks in the center than around the IOs). It then [reduces those samples down](https://github.com/verilog-to-routing/vtr-verilog-to-routing/blob/master/vpr/src/place/timing_place_lookup.cpp#L498-L503) to make the delay matrix.

Most FPGA architectures are sufficiently uniform that keeping separate delay matrices for different block types is actually not that beneficial.

> As a preliminary suggestion, does it make sense to use a nearest neighbor approach to fill "holes" in the delay matrix?

I believe this is [already the case](https://github.com/verilog-to-routing/vtr-verilog-to-routing/blob/master/vpr/src/place/timing_place_lookup.cpp#L594).

> Idea for a better delay matrix sampling: Sample 1 block of each type at min(x) / min(y) and max(x) / max(y)
> This ensures delay matrix coverage from both a spatial standpoint and a routing variety standpoint.

Yep, something like that seems reasonable from a coverage standpoint. However its still probably a good idea to sample from some mid-grid locations to cover the common cases more thoroughly.

litghost, 2019-05-15:
> I believe this is [already the case](https://github.com/verilog-to-routing/vtr-verilog-to-routing/blob/master/vpr/src/place/timing_place_lookup.cpp#L594).

That is true for EMPTY, but not for UNINITIALIZED_DELTA:
https://github.com/verilog-to-routing/vtr-verilog-to-routing/blob/master/vpr/src/place/timing_place_lookup.cpp#L605

closed 2025-05-15 by github-actions[bot]