verilog-to-routing/vtr-verilog-to-routing #569: docs: Fix the missing reference blocks

mithro, 2019-05-16:
When generating the VtR docs at the moment the following issues are output;
```
verilog-to-routing/vtr-verilog-to-routing/doc/src/dev/building.md:205: WARNING: None:any reference target not found: cmake/toolchains/mingw-linux-cross-compile-to-windows.cmake
verilog-to-routing/vtr-verilog-to-routing/doc/src/dev/contributing.md:9: WARNING: None:any reference target not found: SUPPORT
verilog-to-routing/vtr-verilog-to-routing/doc/src/dev/contributing.md:21: WARNING: None:any reference target not found: SUPPORT
verilog-to-routing/vtr-verilog-to-routing/doc/src/dev/contributing.md:122: WARNING: None:any reference target not found: LICENSE
verilog-to-routing/vtr-verilog-to-routing/doc/src/dev/developing.md:1715: WARNING: None:any reference target not found: ./dev/external_subtrees.py
```

These reference targets should probably be fixed.

kmurray, 2019-05-16:
@tinayang3024 Can you look into this?

tinayang3024, 2019-05-16:
Yes, I will work on this.

mithro, 2019-05-16:
Just FYI I'm already looking at this. Just logged it so I didn't forget :-)

kmurray, 2019-09-16:
Should be fixed with #943

closed 2019-09-16 by kmurray