verilog-to-routing/vtr-verilog-to-routing #572: ODIN_II: verilog_preprocessor warnings and error

jeanlego, 2019-05-16:
@hsoontie can look at this?

## Odin compile with warning when compiling using cmake and GNU Make

odin will throw compile warnings for the verilog preprocessor when compiling with cmake and gnu make

## Odin errors on this file at the preprocessor

#### matmul.v
```
`define MY_DWIDTH 16
`define MY_AWIDTH 4
`define MY_MEM_SIZE 16

module matrix_multiplication (
	input clk, 
	input reset, 
	input we, 
	input start, 
	input [`MY_DWIDTH-1:0] data_in, 
	output [`MY_DWIDTH-1:0] data_out
); 

	wire [`MY_DWIDTH-1:0] mat_A;  

	wire [`MY_AWIDTH-1:0] i;
	wire [`MY_AWIDTH-1:0] j;
	reg [`MY_AWIDTH-1:0] addr_in;  

	//Local (distributed) memory
	reg [`MY_DWIDTH-1:0] matrixA[`MY_AWIDTH-1:0][`MY_AWIDTH-1:0];

	// BRAM matrix A  
	single_port_ram matrix_A_u (
		.clk	( clk ),
		.we		( we ),
		.data	( data_in ),
		.out	( mat_A ),
		.addr	( addr_in )
	);

	always @(posedge clk or posedge reset)  
	begin  
		if(reset && !start) begin  
			addr_in <= 0;  
		end  
		else begin  
			if(addr_in<15)
				addr_in = addr_in + 1;  

			i = addr_in >> 2;
			j = addr_in & 4'b0011 ;

			matrixA[i][j] = mat_A ;  
			data_out = mat_A;
		end  
	end  
			 
 endmodule

```

#### gdb --args ./odin_II -V regression_test/benchmark/_bug/matmul.v      
```
GNU gdb (GDB) 8.2.1
Copyright (C) 2018 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "x86_64-pc-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from ./odin_II...done.
(gdb) r
Starting program: /workspace/vtr-verilog-to-routing/ODIN_II/odin_II -V regression_test/benchmark/_bug/matmul.v
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/usr/lib/libthread_db.so.1".
--------------------------------------------------------------------
Welcome to ODIN II version 0.1 - the better High level synthesis tools++ targetting FPGAs (mainly VPR)
Email: jamieson.peter@gmail.com and ken@unb.ca for support issues

--------------------------------------------------------------------
High-level synthesis Begin
Parser starting - we'll create an abstract syntax tree. Note this tree can be viewed using Grap Viz (see documentation)
Preprocessing verilog.
terminate called after throwing an instance of 'std::out_of_range'
  what():  basic_string::substr: __pos (which is 18446744073709551615) > this->size() (which is 2)

Program received signal SIGABRT, Aborted.
0x00007ffff7acf82f in raise () from /usr/lib/libc.so.6
(gdb) bt
#0  0x00007ffff7acf82f in raise () from /usr/lib/libc.so.6
#1  0x00007ffff7aba672 in abort () from /usr/lib/libc.so.6
#2  0x00007ffff7e4958e in __gnu_cxx::__verbose_terminate_handler () at /build/gcc/src/gcc/libstdc++-v3/libsupc++/vterminate.cc:95
#3  0x00007ffff7e4fe0a in __cxxabiv1::__terminate (handler=<optimized out>) at /build/gcc/src/gcc/libstdc++-v3/libsupc++/eh_terminate.cc:47
#4  0x00007ffff7e4fe67 in std::terminate () at /build/gcc/src/gcc/libstdc++-v3/libsupc++/eh_terminate.cc:57
#5  0x00007ffff7e500bc in __cxxabiv1::__cxa_throw (obj=obj@entry=0x555555870e40, tinfo=0x7ffff7f42018 <typeinfo for std::out_of_range>, dest=0x7ffff7e65b90 <std::out_of_range::~out_of_range()>) at /build/gcc/src/gcc/libstdc++-v3/libsupc++/eh_throw.cc:95
#6  0x00007ffff7e4b7aa in std::__throw_out_of_range_fmt (__fmt=__fmt@entry=0x7ffff7f03ed8 "%s: __pos (which is %zu) > this->size() (which is %zu)") from /usr/lib/libstdc++.so.6
#7  0x00007ffff7eeb4bd in std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::_M_check (__s=0x7ffff7f078b5 "basic_string::substr", __pos=<optimized out>, this=<optimized out>)
    at /build/gcc/src/gcc-build/x86_64-pc-linux-gnu/libstdc++-v3/include/bits/basic_string.h:318
#8  std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::substr (this=<optimized out>, __pos=<optimized out>, __n=<optimized out>) at /build/gcc/src/gcc-build/x86_64-pc-linux-gnu/libstdc++-v3/include/bits/basic_string.h:2826
#9  0x00005555556710ae in find_substring[abi:cxx11](char*, char const*, int) (src=0x7fffffff9ad0 ");", sKey=0x5555557e5d76 "module", flag=2) at ../ODIN_II/SRC/odin_util.cpp:778
#10 0x00005555556d5df6 in format_verilog_file (source=0x555555874610) at ../ODIN_II/SRC/verilog_preprocessor.cpp:732
#11 0x00005555556d4ec8 in veri_preproc_bootstraped (original_source=0x55555586fe70, preproc_producer=0x5555558703e0, current_include=0x55555586dbc0) at ../ODIN_II/SRC/verilog_preprocessor.cpp:385
#12 0x00005555556d4c63 in veri_preproc (source=0x55555586fe70) at ../ODIN_II/SRC/verilog_preprocessor.cpp:323
#13 0x000055555569cbf8 in parse_to_ast () at ../ODIN_II/SRC/parse_making_ast.cpp:137
#14 0x00005555556549f2 in synthesize_verilog () at ../ODIN_II/SRC/odin_ii.cpp:110
#15 0x0000555555654daf in start_odin_ii (argc=3, argv=0x7fffffffdea8) at ../ODIN_II/SRC/odin_ii.cpp:254
#16 0x000055555565458b in main ()
```

hsoontie, 2019-05-16:
I just tried this on my own computer and this error didn't come up. Are you using an old version of verilog_preprocessor.cpp? I think on line 732, a few commits ago, there is a call of find_substring that is referenced in the error message.

jeanlego, 2019-05-17:
@hsoontie I've updated the issue

hsoontie, 2019-05-23:
@jeanlego This should be okay to be closed, I think!

closed 2019-05-23 by jeanlego