verilog-to-routing/vtr-verilog-to-routing #579: Odin II Crashes While Processing Nyuzi Processor Project

suyang5, 2019-05-17:
While running Odin II on Nyuzi Processor Project, an error of `std::out_of_range` is incurred.

#### Expected Behaviour

It is expected to convert the entire verilog project into a .blif format file.

#### Current Behaviour

The program terminates during the process with an `std::out_of_range` exception thrown.

#### Possible Solution

#### Steps to Reproduce

[nyuziODIN.tar.gz](https://github.com/verilog-to-routing/vtr-verilog-to-routing/files/3191670/nyuziODIN.tar.gz)
1. Extract nyuziODIN.tar.gz
2. Change the directory in terminal to the extracted folder 
3. enter command `./odin_II -c nyuziVTR.xml`
4. The error message will be shown in the terminal

#### Context

I am trying to push the Nyuzi Processor through the VTR flow.

#### Your Environment

* VTR revision used: Odin_II v 0.1
* Operating System and version: Ubuntu 14.04.5 LTS

jeanlego, 2019-05-17:
sadly this appears to be system Verilog code [github.com/jbush001/NyuziProcessor](https://github.com/jbush001/NyuziProcessor).

I've tried doing an automated conversion tool [github.com/zachjs/sv2v](https://github.com/zachjs/sv2v) to see if that could handle it but AFAIK it contains a lot of unsynthesizable syntaxes. `sv2v` could not handle it.

It would be definitely interesting to see this gets pushed through at some point and we'll keep it in mind as we proceed onto improving Odin support.

vaughnbetz, 2019-05-17:
Hi @jeanlego 
It would be good to at least give an error and a line number instead of an out of range exception.  Is that doable?

jeanlego, 2019-05-19:
I'll try to make that happen thanks :)

jeanlego, 2019-06-22:
Odin will now display an appropriate error when trying to read an unsupported file type (non Verilog)

We don't intend on supporting system Verilog to the level this needs in the near future, I am moving this to a feature request

closed 2025-05-15 by github-actions[bot]