verilog-to-routing/vtr-verilog-to-routing #582: Import PicoSoC as a VtR benchmark

mithro, 2019-05-17:
The [SymbiFlow project](https://symbiflow.github.io) is using [PicoSoC](https://github.com/cliffordwolf/picorv32/tree/master/picosoc). This is a "Cortex M0" style 32bit RISC-V processor + supporting infrastructure (XIP SPI controller, UART, etc).

The [SymbiFlow Architecture Definitions repository](https://github.com/SymbiFlow/symbiflow-arch-defs) is already running PicoSoC through VtR with our own architecture files both for Xilinx Series 7 and Lattice iCE40. You can [find a copy of the exact things we are testing here](https://github.com/SymbiFlow/symbiflow-arch-defs/tree/master/tests/9-soc/picosoc).

It would be good if this benchmark was included in the normal VtR benchmarks.

mithro, 2019-05-17:
FYI - @acomodi - Do you want to take this on, or is it better for @vaughnbetz to get one of his students to look at this?

mithro, 2019-05-17:
One potential issue is that most benchmarks in VtR are using ODIN-II, while SymbiFlow uses Yosys.

vaughnbetz, 2019-05-18:
If @acomodi can try pushing it through Odin-II that would be great.  If there's no error please we can add it to the VTR benchmarks.  If there is an error, please file an issue on it. If this doesn't go through Odin-II and there's no immediate fix, we could still add it as a benchmark, and use yosys for synthesis. 

Once we know what flows this benchmark goes through, and we can figure out how to add it.  If it's a single file and it goes through Odin-II we can just add it to /vtr_flow/benchmarks/verilog and the appropriate regressions like vtr_flow/tasks/regression_tests/vtr_reg_nightly/vtr_reg_qor_chain/config/config.txt (add to circuit_list_add).  If it only goes through Yosys and/or has multiple files we don't want to concantenate we can make a new regression to run it through its own flow.

vaughnbetz, 2019-05-18:
Adding @suyang5 as he is currently pushing new benchmarks through the flow, so he'll be interested in how this is done, and may also help out.

jeanlego, 2019-05-18:
On the Odin side of things, we might be able to get most of picoSoC through at first glance, I'll get someone on the ' inout '. And there's a couple of PR waiting to be merged. Hopefully that should get it in!

acomodi, 2019-05-20:
@vaughnbetz @mithro Sure, I can take a look at this.

mithro, 2019-05-22:
@acomodi You will probably need help from @jeanlego and his team to get this going. When you have a WIP pull request, please publish it somewhere so they can check the ODIN-II side.

acomodi, 2019-05-22:
@mithro Ok, I started to look into this today. When I have something reasonable to push I'll open the PR.
I have a question regarding this though that I did not fully understand: do we want to import also the Artix7 and iCE40 archs and use them in the benchmarks, is that right?

vaughnbetz, 2019-05-22:
I'd consider that a separate issue.  We'd like to be able to use this benchmark with multiple architectures. 
 So:
#1: incorporate this benchmark, and regtest it through the flow targeting one of our favourite architectures, the same one used in the current vtr_flow/tasks/regression_tests/vtr_reg_nightly/vtr_reg_qor_chain/config/config.txt

#2: Add in the Artix7 and iCE40 architectures and make a regtest targeting this same benchmark at them.  I am sure symbiflow already has such a regtest, so the question is whether we should also have one in the vtr source tree and continuous integration; seems like having one makes sense so we see right away if someone breaks these architectures.

closed 2025-05-15 by github-actions[bot]