verilog-to-routing/vtr-verilog-to-routing #584: Import MuraxSoC as a VtR benchmark

mithro, 2019-05-17:
The [SymbiFlow project](https://symbiflow.github.io) is using [MuraxSoC](https://github.com/cliffordwolf/picorv32/tree/master/picosoc). This is the VexRISC-V configured as a "Cortex M0" style 32bit RISC-V processor + supporting infrastructure (XIP SPI controller, UART, etc).

The [SymbiFlow Architecture Definitions repository](https://github.com/SymbiFlow/symbiflow-arch-defs) is already running PicoSoC through VtR with our own architecture files both for Xilinx Series 7 and Lattice iCE40. You can [find a copy of the exact things we are testing here](https://github.com/SymbiFlow/symbiflow-arch-defs/tree/master/tests/9-soc/murax).

It would be good if this benchmark was included in the normal VtR benchmarks.

mithro, 2019-05-17:
FYI - @acomodi - Do you want to take this on, or is it better for @vaughnbetz to get one of his students to look at this?

mithro, 2019-05-17:
One potential issue is that most benchmarks in VtR are using ODIN-II, while SymbiFlow uses Yosys.

vaughnbetz, 2019-05-18:
Adding @suyang5 to this discussion too. My comments are the same as on the earlier benchmark.

closed 2025-05-15 by github-actions[bot]