verilog-to-routing/vtr-verilog-to-routing #589: Odin: Multiplier of size 1 not working, segfault

jeanlego, 2019-05-20:
@MohamedEldafrawy more information on this?

I've put your changes on a branch `odin_multipliers_fix` onto which you can make a PR that I can merge later on.

aarongraham9, 2019-05-20:
@MohamedEldafrawy; Please see branch [master_odin_cmath_fix](https://github.com/verilog-to-routing/vtr-verilog-to-routing/commits/master_odin_cmath_fix) for how I fixed the compile error.

aarongraham9, 2019-05-20:
I brought my fix over to @jeanlego's `odin_multipliers_fix` branch.

MohamedEldafrawy, 2019-05-20:
@aarongraham9 do you have any idea why would the code compile successfully on my machine and fail on Travis?

MohamedEldafrawy, 2019-05-20:
> @MohamedEldafrawy more information on this?
> 
> I've put your changes on a branch `odin_multipliers_fix` from which you can make a PR that I can merge after Travis gives the thumbs up since this might affect our regression tests results.
> 
> Can you provide a Verilog file to reproduce? We'll try to see the root cause and make sure soft-logic also works

So, before this fix 10bea8427aa7e87131990ce67e1fa18098fefc42 if the architecture had a hard multiplier, a soft multiplier will never be implemented regardless of the minimum hard multiplier size provided when running odin. So, the only way to reproduce this behavior before the fix in 10bea8427aa7e87131990ce67e1fa18098fefc42 is to use an architecture that doesn't have a hard multiplier. Here is the vtr flagship architecture after removing the multiplier from it, a Verilog file and odin configuration file to reproduce the problem on the current master [mult_by_1.zip](https://github.com/verilog-to-routing/vtr-verilog-to-routing/files/3199509/mult_by_1.zip)
When running this test through Odin it will terminal at this point https://github.com/verilog-to-routing/vtr-verilog-to-routing/blob/4fcc1bc903862b7bde201c3954869e21415a649c/ODIN_II/SRC/multipliers.cpp#L209
Before this commit 726e5e748788159fe1ca8f0a24d7959c46b8de55, the code I added implements soft multipliers using AND gates and balanced adder tree -- if the architecture has hard adders. In my code, I also added an assert statement for the same condition (one of the operands is 1 bit wide). I have tried to make it work but I have faced two problems:

1. When the soft multiplier is coming directly from the Verilog synthesis meaning we have a small multiplier (7x1) and we want to implement that in soft logic. The input to this function https://github.com/verilog-to-routing/vtr-verilog-to-routing/blob/8ae8064f529631c7f07c74de2576132b67d03dd8/ODIN_II/SRC/multipliers.cpp#L1216 will be a multiplier that has port "a" 7 bits wide and port "b" 1 bit wide and an output port which is 8 bits wide. However, in this case the output should be only 7 bits wide. This unconnected output pin propagate and causes a segmentation fault. I believe that if the output port width is set to 7 and fed to this function it should work fine -- after removing the assert statement.

2. The second problem is when splitting a large multiplier that doesn't fit in the hard multiplier block available in the architecture into two or more multipliers using the code in https://github.com/verilog-to-routing/vtr-verilog-to-routing/blob/4fcc1bc903862b7bde201c3954869e21415a649c/ODIN_II/SRC/multipliers.cpp#L716 https://github.com/verilog-to-routing/vtr-verilog-to-routing/blob/4fcc1bc903862b7bde201c3954869e21415a649c/ODIN_II/SRC/multipliers.cpp#L822 https://github.com/verilog-to-routing/vtr-verilog-to-routing/blob/4fcc1bc903862b7bde201c3954869e21415a649c/ODIN_II/SRC/multipliers.cpp#L898 and one of those has a port that is 1 bit wide. Probably the width of the additions that is combining the output of those multipliers is affected causing also a segmentation fault termination.

aarongraham9, 2019-05-21:
It broke on my local build environment as well as Travis CI. Not sure why it builds for you @MohamedEldafrawy. I don't see how it could build at all when the header is missing an include.

MohamedEldafrawy, 2019-05-21:
Thanks @aarongraham9 I think it's a compiler version issue. Probably newer compilers can include some libraries (cmath) implicitly when including others (algorithm).

jeanlego, 2019-05-22:
Just to be sure I'm getting this correctly, PR #590 is a workaround for this issue but a proper fix is still needed.
if so, I'll make sure someone looks at it, if not myself, in the very near future.

:) thanks for raising this issue!

MohamedEldafrawy, 2019-05-22:
You are definitely welcome :)

That's not really what PR #590 is about. So, one of the things done in PR #590 is fixing the condition for implementing multipliers using soft logic if they are smaller than a value provide by the user -- in the vtr regression tests this value is set to 3. Before this fix all multiplier were implemented using hard multipliers. Therefore, this issue wasn't there, since a hard multiplier has a fixed number of input/output pins therefore, counting the number of outputs in a wrong way wasn't affecting the code. Since now the condition is fixed, sometimes we will end up in a scenario where a multiplier is implemented using soft logic with one or both of it's inout ports having a size of 1. For soft multiplier each output pin is connected to a logic element (LUT or adder) therefore having an unconnected pin in the output causes a segmentation fault later on. 

So, basically since PR #590 fixed the condition this work around was added to avoid odin from crashing -- it's also important to notice that this work around is aligned with how odin used to do before the PR since all multipliers were implelemted using hard multipliers regardless of their size.

Other than that PR #590 has also the following:
1. Fixed the logic of splitting large multipliers that cannot fit into the available hard multipliers into smaller ones.
2. Added a way of implementing soft multipliers using AND gates and a balanced adder tree. Instead of LUTs only previously.

jeanlego, 2019-05-22:
thanks for this :) 

merged #590

MohamedEldafrawy, 2019-09-02:
Hi @jeanlego,

Any progress on this issue?

jeanlego, 2019-11-01:
@georgkrylov this might be of interest to you

closed 2025-05-15 by github-actions[bot]