verilog-to-routing/vtr-verilog-to-routing #595: Travis build matrix with gcc [4 to 8]

jeanlego, 2019-05-23:
#### Proposed Behaviour
It seems that some issues creep in the buildbot regression because of gcc versions.

Travis offers build matrix that can run in parralel and could probably catch thos early
https://docs.travis-ci.com/user/build-matrix/

I was thinkng this might be a good idea for our use case.

I'll whip up something if there's interest.

kmurray, 2019-05-23:
FYI, the VTR Buildbot does already check that VTR compiles [with various GCC & Clang versions](http://builds.verilogtorouting.org:8080/builders/build_configs/builds/760/steps/sweep%20build%20configurations/logs/stdio). It also checks that the code compiles correctly with various compilation flag combinations. See [the script](https://github.com/verilog-to-routing/vtr-verilog-to-routing/blob/master/sweep_build_configs.py) for details.

mithro, 2019-05-31:
See #625

mithro, 2019-05-31:
Example implementation can be found on my `build-compat` branch which [builds on top of the pull request #619]( https://github.com/verilog-to-routing/vtr-verilog-to-routing/pull/619).

Example output can be found at https://travis-ci.org/mithro/vtr-verilog-to-routing/builds/539520874 and seen below;

[image]

A quick look it seems like;
 - [x] GCC 5 works
 - [x] GCC 6 works
 - [ ] GCC 7 fails with issue in https://github.com/verilog-to-routing/vtr-verilog-to-routing/issues/628
 - [ ] GCC 8 fails with issue in https://github.com/verilog-to-routing/vtr-verilog-to-routing/issues/628
 - [ ] clang-3.9 currently a WIP
 - [ ] clang-6.0 currently a WIP
 - [ ] clang-8 fails with a similar issue in the linkter to https://github.com/verilog-to-routing/vtr-verilog-to-routing/issues/628

kmurray, 2019-05-31:
Interesting. [On buildbot](http://builds.verilogtorouting.org:8080/builders/build_configs/builds/774) we build the following successfully:
* gcc: 4.9, 5, 6, 7, 8
* clang: 3.6, 3.8
* mingw-64 (gcc cross-compile to windows): 5

Any ideas about what's different between those environments?

mithro, 2019-06-02:
@kmurray The only inconsistency there is gcc 8? Travis doesn't run the older clang or gcc 4.9

kmurray, 2019-07-23:
Resolved with #818

closed 2019-07-23 by kmurray