verilog-to-routing/vtr-verilog-to-routing #597: ODIN: preprocessor buffer too small, >10000 col for a single line in titan benchmark

j-b-1-7, 2019-05-23:
smithwaterman.v fails at the preprocessor due to the buffer being too small

[smithwaterman.v.txt](https://github.com/verilog-to-routing/vtr-verilog-to-routing/files/3210129/smithwaterman.v.txt)

line 73964 has >10000 character for a single line.

kmurray, 2019-05-23:
I recall fixing similar issues in VPR previously. Generally it's a good idea to have no hard-coded limit, and instead dynamically grow the buffer if needed.

ademmings, 2019-05-29:
[preprocessor_charlimit.v.txt](https://github.com/verilog-to-routing/vtr-verilog-to-routing/files/3234332/preprocessor_charlimit.v.txt)

```
module BS_LINE(
	ap_CS_fsm,
	query_empty_n,
	ap_const_logic_0,
	ap_ST_st1_fsm_1,
	ap_ST_st2_fsm_2,
	ap_ST_st3_fsm_3,
	ap_ST_st4_fsm_4,
	ap_ST_st5_fsm_5,
	ap_ST_st6_fsm_6,
	ap_ST_st7_fsm_7,
	ap_ST_st8_fsm_8,
	ap_ST_st9_fsm_9,
	ap_ST_st10_fsm_10,
	ap_ST_st11_fsm_11,
	ap_ST_st12_fsm_12,
	ap_ST_st13_fsm_13,
	ap_ST_st14_fsm_14,
	ap_ST_st15_fsm_15,
	ap_ST_st16_fsm_16,
	ap_ST_st17_fsm_17,
	ap_ST_st18_fsm_18,
	ap_ST_st19_fsm_19,
	ap_ST_st20_fsm_20,
	ap_ST_st21_fsm_21,
	ap_ST_st22_fsm_22,
	ap_ST_st23_fsm_23,
	ap_ST_st24_fsm_24,
	ap_ST_st25_fsm_25,
	ap_ST_st26_fsm_26,
	ap_ST_st27_fsm_27,
	ap_ST_st28_fsm_28,
	ap_ST_st29_fsm_29,
	ap_ST_st30_fsm_30,
	ap_ST_st31_fsm_31,
	ap_ST_st32_fsm_32,
	ap_ST_st33_fsm_33,
	ap_ST_st34_fsm_34,
	ap_ST_st35_fsm_35,
	ap_ST_st36_fsm_36,
	ap_ST_st37_fsm_37,
	ap_ST_st38_fsm_38,
	ap_ST_st39_fsm_39,
	ap_ST_st40_fsm_40,
	ap_ST_st41_fsm_41,
	ap_ST_st42_fsm_42,
	ap_ST_st43_fsm_43,
	ap_ST_st44_fsm_44,
	ap_ST_st45_fsm_45,
	ap_ST_st46_fsm_46,
	ap_ST_st47_fsm_47,
	ap_ST_st48_fsm_48,
	ap_ST_st49_fsm_49,
	ap_ST_st50_fsm_50,
	ap_ST_st51_fsm_51,
	ap_ST_st52_fsm_52,
	ap_ST_st53_fsm_53,
	ap_ST_st54_fsm_54,
	ap_ST_st55_fsm_55,
	ap_ST_st56_fsm_56,
	ap_ST_st57_fsm_57,
	ap_ST_st58_fsm_58,
	ap_ST_st59_fsm_59,
	ap_ST_st60_fsm_60,
	ap_ST_st61_fsm_61,
	ap_ST_st62_fsm_62,
	ap_ST_st63_fsm_63,
[... 192 lines cut ...]
	input ap_ST_st121_fsm_121 ;
	input ap_ST_st122_fsm_122 ;
	input ap_ST_st123_fsm_123 ;
	input ap_ST_st124_fsm_124 ;
	input ap_ST_st125_fsm_125 ;
	input ap_ST_st126_fsm_126 ;
	input ap_ST_st127_fsm_127 ;

	/* output declaration */
	output ap_cond148_0;

	/* crazy evaluation */

	always @ (ap_CS_fsm or query_empty_n)
	begin
	    if ((((((((((((((((((((((((((((((((((((((((((((((((((((((((((((((((((((((((((((((((((((((((((((((((((((((((((((((((((((((((((((((((((ap_ST_st128_fsm_128 == ap_CS_fsm) & ~(query_empty_n == ap_const_logic_0)) | (~(query_empty_n == ap_const_logic_0) & (ap_ST_st1_fsm_1 == ap_CS_fsm))) | (~(query_empty_n == ap_const_logic_0) & (ap_ST_st2_fsm_2 == ap_CS_fsm))) | (~(query_empty_n == ap_const_logic_0) & (ap_ST_st3_fsm_3 == ap_CS_fsm))) | (~(query_empty_n == ap_const_logic_0) & (ap_ST_st4_fsm_4 == ap_CS_fsm))) | (~(query_empty_n == ap_const_logic_0) & (ap_ST_st5_fsm_5 == ap_CS_fsm))) | (~(query_empty_n == ap_const_logic_0) & (ap_ST_st6_fsm_6 == ap_CS_fsm))) | (~(query_empty_n == ap_const_logic_0) & (ap_ST_st7_fsm_7 == ap_CS_fsm))) | (~(query_empty_n == ap_const_logic_0) & (ap_ST_st8_fsm_8 == ap_CS_fsm))) | (~(query_empty_n == ap_const_logic_0) & (ap_ST_st9_fsm_9 == ap_CS_fsm))) | (~(query_empty_n == ap_const_logic_0) & (ap_ST_st10_fsm_10 == ap_CS_fsm))) | (~(query_empty_n == ap_const_logic_0) & (ap_ST_st11_fsm_11 == ap_CS_fsm))) | (~(query_empty_n == ap_const_logic_0) & (ap_ST_st12_fsm_12 == ap_CS_fsm))) | (~(query_empty_n == ap_const_logic_0) & (ap_ST_st13_fsm_13 == ap_CS_fsm))) | (~(query_empty_n == ap_const_logic_0) & (ap_ST_st14_fsm_14 == ap_CS_fsm))) | (~(query_empty_n == ap_const_logic_0) & (ap_ST_st15_fsm_15 == ap_CS_fsm))) | (~(query_empty_n == ap_const_logic_0) & (ap_ST_st16_fsm_16 == ap_CS_fsm))) | (~(query_empty_n == ap_const_logic_0) & (ap_ST_st17_fsm_17 == ap_CS_fsm))) | (~(query_empty_n == ap_const_logic_0) & (ap_ST_st18_fsm_18 == ap_CS_fsm))) | (~(query_empty_n == ap_const_logic_0) & (ap_ST_st19_fsm_19 == ap_CS_fsm))) | (~(query_empty_n == ap_const_logic_0) & (ap_ST_st20_fsm_20 == ap_CS_fsm))) | (~(query_empty_n == ap_const_logic_0) & (ap_ST_st21_fsm_21 == ap_CS_fsm))) | (~(query_empty_n == ap_const_logic_0) & (ap_ST_st22_fsm_22 == ap_CS_fsm))) | (~(query_empty_n == ap_const_logic_0) & (ap_ST_st23_fsm_23 == ap_CS_fsm))) | (~(query_empty_n == ap_const_logic_0) & (ap_ST_st24_fsm_24 == ap_CS_fsm))) | (~(query_empty_n == ap_const_logic_0) & (ap_ST_st25_fsm_25 == ap_CS_fsm))) | (~(query_empty_n == ap_const_logic_0) & (ap_ST_st26_fsm_26 == ap_CS_fsm))) | (~(query_empty_n == ap_const_logic_0) & (ap_ST_st27_fsm_27 == ap_CS_fsm))) | (~(query_empty_n == ap_const_logic_0) & (ap_ST_st28_fsm_28 == ap_CS_fsm))) | (~(query_empty_n == ap_const_logic_0) & (ap_ST_st29_fsm_29 == ap_CS_fsm))) | (~(query_empty_n == ap_const_logic_0) & (ap_ST_st30_fsm_30 == ap_CS_fsm))) | (~(query_empty_n == ap_const_logic_0) & (ap_ST_st31_fsm_31 == ap_CS_fsm))) | (~(query_empty_n == ap_const_logic_0) & (ap_ST_st32_fsm_32 == ap_CS_fsm))) | (~(query_empty_n == ap_const_logic_0) & (ap_ST_st33_fsm_33 == ap_CS_fsm))) | (~(query_empty_n == ap_const_logic_0) & (ap_ST_st34_fsm_34 == ap_CS_fsm))) | (~(query_empty_n == ap_const_logic_0) & (ap_ST_st35_fsm_35 == ap_CS_fsm))) | (~(query_empty_n == ap_const_logic_0) & (ap_ST_st36_fsm_36 == ap_CS_fsm))) | (~(query_empty_n == ap_const_logic_0) & (ap_ST_st37_fsm_37 == ap_CS_fsm))) | (~(query_empty_n == ap_const_logic_0) & (ap_ST_st38_fsm_38 == ap_CS_fsm))) | (~(query_empty_n == ap_const_logic_0) & (ap_ST_st39_fsm_39 == ap_CS_fsm))) | (~(query_empty_n == ap_const_logic_0) & (ap_ST_st40_fsm_40 == ap_CS_fsm))) | (~(query_empty_n == ap_const_logic_0) & (ap_ST_st41_fsm_41 == ap_CS_fsm))) | (~(query_empty_n == ap_const_logic_0) & (ap_ST_st42_fsm_42 == ap_CS_fsm))) | (~(query_empty_n == ap_const_logic_0) & (ap_ST_st43_fsm_43 == ap_CS_fsm))) | (~(query_empty_n == ap_const_logic_0) & (ap_ST_st44_fsm_44 == ap_CS_fsm))) | (~(query_empty_n == ap_const_logic_0) & (ap_ST_st45_fsm_45 == ap_CS_fsm))) | (~(query_empty_n == ap_const_logic_0) & (ap_S [... cut ...] ap_const_logic_0) & (ap_ST_st124_fsm_124 == ap_CS_fsm))) | (~(query_empty_n == ap_const_logic_0) & (ap_ST_st125_fsm_125 == ap_CS_fsm))) | (~(query_empty_n == ap_const_logic_0) & (ap_ST_st126_fsm_126 == ap_CS_fsm))) | (~(query_empty_n == ap_const_logic_0) & (ap_ST_st127_fsm_127 == ap_CS_fsm)))) begin
		ap_cond148_0 = ap_const_logic_1;
	    end else begin
		ap_cond148_0 = ap_const_logic_0;
	    end
	end

endmodule
```
here's the most important part of the smithwaterman.v file, for testing purposes

jeanlego, 2019-06-22:
Fixed
Closing

closed 2019-06-22 by jeanlego