verilog-to-routing/vtr-verilog-to-routing #598: How to specify combinational delay for class="lut" objects?

mithro, 2019-05-23:
How do you specify the combinational delay for a LUT which has a `class="LUT"` attribute? I can't seem to figure it out, both using the `port_name` and `port_class` values in the `<delay_constant />` tag end up with the same error.

#### Expected Behaviour

You can provide a delay constant.

#### Current Behaviour

If I try and use the `name` values for the `delay_constant` tag as shown below I get 
```
lut.arch.merged.xml:37 Failed to find port 'I' on 'LUT' for combinational delay annotation
```
```xml
      <pb_type blif_model=".names" class="lut" name="LUT" num_pb="1">
        <input name="I" num_pins="4" port_class="lut_in"/>
        <output name="O" num_pins="1" port_class="lut_out"/>
        <delay_constant in_port="LUT.I" max="0.329e-9" out_port="LUT.O"/>
      </pb_type>
```

If I try and use the `port_class` values for the `delay_constant` tag as shown below I get 
```
lut.arch.merged.xml:37 Failed to find port 'lut_in' on 'LUT' for combinational delay annotation
```
```xml
      <pb_type blif_model=".names" class="lut" name="LUT" num_pb="1">
        <input name="I" num_pins="4" port_class="lut_in"/>
        <output name="O" num_pins="1" port_class="lut_out"/>
        <delay_constant in_port="LUT.lut_in" max="0.329e-9" out_port="LUT.lut_out"/>
      </pb_type>
```

#### Possible Solution

If I instead use a `blif_model=".subckt lut"` as shown below it seems to work;
```xml
      <pb_type blif_model=".subckt lut" name="LUT" num_pb="1">
        <input name="I" num_pins="4"/>
        <output name="O" num_pins="1"/>
        <delay_constant in_port="LUT.I" max="0.329e-9" out_port="LUT.O"/>
      </pb_type>
```

#### Steps to Reproduce

Full arch.xml provided below;
```xml
<?xml version="1.0"?>
<!-- set: ai sw=1 ts=1 sta et -->
<architecture xmlns:xi="http://www.w3.org/2001/XInclude">
  <models>
    <model name="lut">
      <input_ports>
        <port name="I" combinational_sink_ports="O"/>
      </input_ports>
      <output_ports>
        <port name="O"/>
      </output_ports>
    </model>
  </models>
  <complexblocklist>
    <!-- input buffer -->
    <pb_type name="IBUF">
      <output name="I" num_pins="1"/>
      <pb_type blif_model=".input" name="PIN" num_pb="1">
        <output name="inpad" num_pins="1"/>
      </pb_type>
      <interconnect>
        <direct input="PIN.inpad" name="P" output="IBUF.I"/>
      </interconnect>
      <pinlocations pattern="custom">
        <loc side="right" xoffset="0" yoffset="0">IBUF.I</loc>
      </pinlocations>
    </pb_type>
    <!-- output buffer -->
    <pb_type name="OBUF">
      <input name="O" num_pins="1"/>
      <pb_type blif_model=".output" name="PIN" num_pb="1">
        <input name="outpad" num_pins="1"/>
      </pb_type>
      <interconnect>
        <direct input="OBUF.O" name="P" output="PIN.outpad"/>
      </interconnect>
      <pinlocations pattern="custom">
        <loc side="left" xoffset="0" yoffset="0">OBUF.O</loc>
      </pinlocations>
    </pb_type>
    <!-- pb_type.xml file will be included here -->
    <pb_type height="4" name="TILE" width="1">
      <pb_type blif_model=".subckt lut" name="LUT" num_pb="1">
        <input name="I" num_pins="4"/>
        <output name="O" num_pins="1"/>
        <delay_constant in_port="LUT.I" max="0.329e-9" out_port="LUT.O"/>
      </pb_type>
      <pinlocations pattern="custom">
        <loc side="left" xoffset="0" yoffset="0">
TILE.I0
TILE.I1
TILE.I2
TILE.I3
</loc>
        <loc side="right" xoffset="0" yoffset="0">
TILE.O
</loc>
        <loc side="left" xoffset="0" yoffset="1">
TILE.I0
TILE.I1
TILE.I2
TILE.I3
</loc>
        <loc side="right" xoffset="0" yoffset="1">
TILE.O
</loc>
        <loc side="left" xoffset="0" yoffset="2">
TILE.I0
TILE.I1
TILE.I2
TILE.I3
</loc>
        <loc side="right" xoffset="0" yoffset="2">
TILE.O
</loc>
        <loc side="left" xoffset="0" yoffset="3">
TILE.I0
TILE.I1
TILE.I2
TILE.I3
</loc>
        <loc side="right" xoffset="0" yoffset="3">
TILE.O
</loc>
      </pinlocations>
      <interconnect>
        <direct input="TILE.I0" name="TILE.I0-LUT.I[0]" output="LUT.I[0]"/>
        <direct input="TILE.I1" name="TILE.I1-LUT.I[1]" output="LUT.I[1]"/>
        <direct input="TILE.I2" name="TILE.I2-LUT.I[2]" output="LUT.I[2]"/>
        <direct input="TILE.I3" name="TILE.I3-LUT.I[3]" output="LUT.I[3]"/>
        <direct input="LUT.O" name="TILE.O-LUT.O" output="TILE.O"/>
      </interconnect>
      <input equivalent="none" name="I0" num_pins="1"/>
      <input equivalent="none" name="I1" num_pins="1"/>
      <input equivalent="none" name="I2" num_pins="1"/>
      <input equivalent="none" name="I3" num_pins="1"/>
      <output equivalent="none" name="O" num_pins="1"/>
    </pb_type>
  </complexblocklist>
  <layout>
    <!-- layout will be auto generated here -->
    <fixed_layout height="6" name="device" width="6">
      <!--
   01234
 0 .....
 1 I.T..
 2 I.T.O
 3 I.T..
 4 I.T..
 5 .....
-->
      <single priority="1" type="IBUF" x="0" y="1"/>
      <single priority="1" type="IBUF" x="0" y="2"/>
      <single priority="1" type="IBUF" x="0" y="3"/>
      <single priority="1" type="IBUF" x="0" y="4"/>
      <single priority="1" type="TILE" x="2" y="1"/>
      <single priority="1" type="OBUF" x="4" y="2"/>
    </fixed_layout>
  </layout>
  <device>
    <sizing R_minW_nmos="6065.520020" R_minW_pmos="18138.500000"/>
    <area grid_logic_tile_area="14813.392"/>
    <connection_block input_switch_name="mux"/>
    <switch_block fs="3" type="wilton"/>
    <chan_width_distr>
      <x distr="uniform" peak="1.000000"/>
      <y distr="uniform" peak="1.000000"/>
    </chan_width_distr>
    <default_fc in_type="frac" in_val="1.0" out_type="frac" out_val="1.0"/>
  </device>
  <switchlist>
    <switch Cin=".77e-15" Cout="4e-15" R="551" Tdel="58e-12" buf_size="27.645901" mux_trans_size="2.630740" name="mux" type="mux"/>
  </switchlist>
  <segmentlist>
    <segment Cmetal="22.5e-15" Rmetal="101" freq="1.000000" length="1" name="local" type="unidir">
      <sb type="pattern">1 1</sb>
      <cb type="pattern">1</cb>
      <mux name="mux"/>
    </segment>
  </segmentlist>
</architecture>
```

mithro, 2019-05-23:
The following works....
```xml
      <pb_type blif_model=".names" class="lut" name="LUT" num_pb="1">
        <input name="in" num_pins="4" port_class="lut_in"/>
        <output name="out" num_pins="1" port_class="lut_out"/>
        <delay_constant in_port="LUT.in" max="0.329e-9" out_port="LUT.out"/>
      </pb_type>
```

kmurray, 2019-05-24:
Yep, looks like some hard coded assumptions in the architecture file parsing & processing.

It would be good for someone to fix that.

mithro, 2019-05-24:
@duck2 - Might be worth looking into.

closed 2025-05-15 by github-actions[bot]