verilog-to-routing/vtr-verilog-to-routing #602: ODIN: Parse error always block with array condition

j-b-1-7, 2019-05-24:
#### Current Behaviour
Parse error due to array condition in always block

### Context
Tried to compile the file carpat.v included in the titan benchmarks, but the array condition in the always block is giving a parse error

#### uname -a     
```                                                                                                                      
Linux JBpc 5.1.2-arch1-1-ARCH #1 SMP PREEMPT Wed May 15 00:09:47 UTC 2019 x86_64 GNU/Linux
```
#### gcc -v
```
Using built-in specs.
COLLECT_GCC=gcc
COLLECT_LTO_WRAPPER=/usr/lib/gcc/x86_64-pc-linux-gnu/8.3.0/lto-wrapper
Target: x86_64-pc-linux-gnu
Configured with: /build/gcc/src/gcc/configure --prefix=/usr --libdir=/usr/lib --libexecdir=/usr/lib --mandir=/usr/share/man --infodir=/usr/share/info --with-bugurl=https://bugs.archlinux.org/ --enable-languages=c,c++,ada,fortran,go,lto,objc,obj-c++ --enable-shared --enable-threads=posix --enable-libmpx --with-system-zlib --with-isl --enable-__cxa_atexit --disable-libunwind-exceptions --enable-clocale=gnu --disable-libstdcxx-pch --disable-libssp --enable-gnu-unique-object --enable-linker-build-id --enable-lto --enable-plugin --enable-install-libiberty --with-linker-hash-style=gnu --enable-gnu-indirect-function --enable-multilib --disable-werror --enable-checking=release --enable-default-pie --enable-default-ssp --enable-cet=auto
Thread model: posix
gcc version 8.3.0 (GCC) 

```
### carpat.v
[carpat_original.v.txt](https://github.com/verilog-to-routing/vtr-verilog-to-routing/files/3217189/carpat_original.v.txt)

```
always@(posedge CCD_MCLK or negedge KEY[3])	
```

#### ./odin_II -V regression_test/benchmark/my_bugs/titan_benchmarks/other_benchmarks/carpat/quartus2_proj/hdl/carpat.v 
```    
--------------------------------------------------------------------
Welcome to ODIN II version 0.1 - the better High level synthesis tools++ targetting FPGAs (mainly VPR)
Email: jamieson.peter@gmail.com and ken@unb.ca for support issues

--------------------------------------------------------------------
High-level synthesis Begin
Parser starting - we'll create an abstract syntax tree. Note this tree can be viewed using Grap Viz (see documentation)
Preprocessing verilog.
error in parsing: (syntax error, unexpected '[', expecting vOR or ')' or ',')::410

```
If the second part is commented out (statements shown below), it will pass through. 
```
always@(posedge CCD_MCLK /*or negedge KEY[3]*/)	
```
```
always@(posedge CCD_MCLK or negedge KEY /*[3]*/ )	
```

jeanlego, 2020-03-31:
@sdamghan can you verify this is still an issue, I think this was resolved. if not it should be a simple fix

sdamghan, 2020-04-02:
@jeanlego It is already solved, we only have inout problem for this Verilog file.

jeanlego, 2020-04-02:
Thanks. Closing

closed 2020-04-02 by jeanlego