verilog-to-routing/vtr-verilog-to-routing #606: Allow some channel segments to have non-consecutive ptc_num values

vaughnbetz, 2019-05-24:
#### Proposed Behaviour

Allow ptc_num values on routing resource nodes (CHANX, CHANY, maybe SINK, etc.) to be non-consecutive at a certain (x,y) location.  

#### Current Behaviour
Symbiflow requires vpr code changes to allow capacity 0 rr_nodes, and must insert those capacity 0 rr_nodes in its rr_graph.

#### Possible Solution

Currently the ptc_num values used on routing resource nodes (CHANX, CHANY, SINK, etc.) must be consecutive for all the wires, nodes, etc. at a certain (x,y) location. This is limiting for some rr-graphs, such as that imported for the Artix 7 devices, so we should update the data structures to remove this restriction.

This consecutive ptc_num behaviour is not required by the router or the rr-graph, but is required by the current code that constructs the rr_node_indices data structure that allows a fast lookup from a certain type of rr-node at a certain (x,y) location back to the rr_node index. This look-up data structure is not used very extensively in the code. It is used during rr-graph construction and might be used for hit testing when someone clicks in the GUI, but not much else.

See the code at https://github.com/verilog-to-routing/vtr-verilog-to-routing/blob/master/vpr/src/route/rr_graph_reader.cpp#L751-L795 for code that will fail if ptc_num is not consecutive.  This code can be fixed by changing the allocation of the vector (indices[CHANY][ix][iy][0]) to instead set its size to the maximum ptc_num at that location + 1.

Basically:

1. Loop through once finding the maximum ptc_num at each (x,y) and storing it for each (x,y) in some variable.

2. Then resize or construct indices[CHANY][ix][iy][0] to have ptc_num(ix,iy) + 1 entries.

3. Initialize all the values in the vector to -1 (OPEN/INVALID) so any ptc_num values that aren't used have an invalid node value (we shouldn't look at them).

4. Then the load loops can run normally, and should never fail the size check.

5. There are not too many points in the code where we use this indices data structure; it's a reverse lookup that lets us find the nodes at a certain location when constructing an rr_graph, and I don't think it's used for much else. There may be a few spots where code needs to be updated to understand that there may not be any rr_node at a certain x,y,ptc_num (i.e. you get 
OPEN/INVALID) for a certain lookup and should skip over it.

#### Context

This change lets us be more flexible in assigning ptc_num values to rr_nodes, which should be particularly helpful for CHANX and CHANY nodes.  The ptc_num value is useful for drawing these nodes (nodes that overlap in length should have different ptc_num values so they don't draw on top of each other).  We also use it during rr-graph construction.  If an external tool like the symbiflow tools was used to create the rr-graph, there is no need to insist that ptc_num values are consecutive with no gaps at a certain (x,y) location, so this change makes vtr more flexible.

mithro, 2019-05-24:
@vaughnbetz Please assign to @litghost and @acomodi rather than myself :-)

vaughnbetz, 2019-05-27:
Sorry about that; I needed to add them as collaborators (I think) before I could assign to them, as I can assign to them now.

acomodi, 2019-09-11:
@vaughnbetz I believe this can be closed, fixed by #648

closed 2019-09-11 by vaughnbetz