verilog-to-routing/vtr-verilog-to-routing #614: VPR cannot find a possible route

acomodi, 2019-05-28:
#### Expected Behaviour

Routing should complete successfully

#### Current Behaviour

The router fails with the following message after the first iteration:
```
 ## Loading routing resource graph took 14.74 seconds (max_rss 2794.0 MiB, delta_rss +35.3 MiB)                                                                                                                    
   RR Graph Nodes: 1425226                                                                                                                                                                                         
   RR Graph Edges: 4631689                                                                                                                                                                                         
 Confirming router algorithm: TIMING_DRIVEN.                                                                                                                                                                       
 ---- ------ ------- ---- ------- ------- ------- ----------------- --------------- -------- ---------- ---------- ---------- ---------- --------                                                                  
 Iter   Time    pres  BBs    Heap  Re-Rtd  Re-Rtd Overused RR Nodes      Wirelength      CPD       sTNS       sWNS       hTNS       hWNS Est Succ                                                                  
       (sec)     fac Updt    push    Nets   Conns                                       (ns)       (ns)       (ns)       (ns)       (ns)     Iter                                                                  
 ---- ------ ------- ---- ------- ------- ------- ----------------- --------------- -------- ---------- ---------- ---------- ---------- --------                                                                  
 Cannot route from SYN-GND.GND[0] (RR node: 113480 type: SOURCE location: (35,26) class: 0) to BLK-TL-SLICEL.AX[0] (RR node: 290922 type: SINK location: (71,34) class: 6) -- no possible path                     
 Failed to route connection from '$false' to '$auto$alumacc.cc:474:replace_alu$6.slice[12].carry' for net '$false' (#2)                                                                                            
 Routing failed.                                                         
```

#### Steps to Reproduce

Logs and files to reproduce: https://filebin.net/nbocf43hgy316tmd
1. compile VTR using following revision: https://github.com/SymbiFlow/vtr-verilog-to-routing/pull/59/commits/a9700f798ee08f8ea03ee0f95d975fe9fa1a0849
2. extract the archive
3. `cd logs/echo`
4. Run:
```
vpr ../arch.unique_pack.xml  ../atom_netlist.cleaned.echo.eblif --device xc7a50t-basys3-test --read_rr_graph ../rr_graph_xc7a50t-basys3_test.rr_graph.real.xml --min_route_chan_width_hint 100 --max_criticality 0.0 --max_router_iterations 500 --routing_failure_predictor off --router_high_fanout_threshold -1 --constant_net_method route --route_chan_width 500 --clock_modeling route --place_algorithm bounding_box --enable_timing_computations off --allow_unrelated_clustering on --clustering_pin_feasibility_filter off --disable_check_route on --echo_file on --pack --place --route 
```

#### Context

The issue showed up after the changes made to solve issue https://github.com/verilog-to-routing/vtr-verilog-to-routing/issues/524.

#### Your Environment

* VTR revision used: https://github.com/SymbiFlow/vtr-verilog-to-routing/pull/59/commits/a9700f798ee08f8ea03ee0f95d975fe9fa1a0849
* Operating System and version: Linux/Ubuntu 18.04
* Compiler version: gcc (Ubuntu 7.3.0-27ubuntu1~18.04) 7.3.0

acomodi, 2019-05-28:
@kmurray

mithro, 2019-05-28:
@litghost - Could you help @acomodi look into this?

kmurray, 2019-05-31:
**Summary**
I've tracked down the cause, which is related to the router is hitting the edge of the `$false` net's bounding box before it actually gets onto what I assume is a global network (connected by shorts).

A work-around for this RR graph is to run with a large bounding box factor to ensure each net's bounding box is the full device (e.g. `--bb_factor 100000`).
This will be slow (from a run-time perspective) but robust.

VPR assumes that a path exists between two points in the FPGA within the (expanded) bounding box of the source and sink.
This is true not necessarily true if the **only** path between those two points is on a global network where the required path leaves the bounding box.

How accurate/realistic is the global network modelled in this architecture? Are all the RR node co-ordinates correct?

**Details**
As noted in the error message, the offending net number is `2`:
```
Failed to route connection from '$false' to '$auto$alumacc.cc:474:replace_alu$6.slice[12].carry' for net '$false' (#2)                                                                                            
```

To see what the router is doing in detail compile with VTR_ENABLE_DEBUG_LOGGING on and run with `--router_debug_net 2`:
```
Net 2 Target 0 (RR node: 169122 type: SINK location: (47,34) class: 6)
  Adding node   113480 to heap from init route tree with cost 0 (RR node: 113480 type: SOURCE location: (35,26) class: 0)
  Routing to 169122 as normal net (BB: 31,22 x 50,40)
  Popping node 113480
    Better cost to 113480
      Setting path costs for assicated node 113480 (from -1 edge -1)
      Expanding node 113480 edge 0 -> 113481
      Expanding to node 113481 (RR node: 113481 type: OPIN location: (35,26) pin: 0 pin_name: SYN-GND.GND[0])
  Popping node 113481
    Better cost to 113481
      Setting path costs for assicated node 113481 (from 113480 edge 0)
      Expanding node 113481 edge 0 -> 993424
      Expanding to node 993424 (RR node: 993424 type: CHANY location: (35,1) <-> (35,156) track: 0 len: 155 longline: 0 seg_type: dummy dir: BI_DIR)
  Popping node 993424
    Better cost to 993424
      Setting path costs for assicated node 993424 (from 113481 edge 0)
  Empty heap (no path found)
Cannot route from SYN-GND.GND[0] (RR node: 113480 type: SOURCE location: (35,26) class: 0) to BLK-TL-SLICEL.AX[0] (RR node: 169122 type: SINK location: (47,34) class: 6) -- no possible path
Failed to route connection from '$false' to '$auto$alumacc.cc:474:replace_alu$6.C[13]' for net '$false' (#2)
Routing failed.
```

Inspecting the RR graph shows that node 934244 connects to 993481 (which connects to many other nodes).
However 993481 is reported to have location (1,155) <-> (147,155), which falls out side the net's bounding box 31,22 x 50,40 and so is not explored.

To work around this run with a large --bb_factor to increase the size of the router's bounding box.
Running with `--bb_factor 100000` (large enough to ensure the bounding box is the full device) I get:
```
Confirming router algorithm: TIMING_DRIVEN.
---- ------ ------- ---- ------- ------- ------- ----------------- --------------- -------- ---------- ---------- ---------- ---------- --------
Iter   Time    pres  BBs    Heap  Re-Rtd  Re-Rtd Overused RR Nodes      Wirelength      CPD       sTNS       sWNS       hTNS       hWNS Est Succ
      (sec)     fac Updt    push    Nets   Conns                                       (ns)       (ns)       (ns)       (ns)       (ns)     Iter
---- ------ ------- ---- ------- ------- ------- ----------------- --------------- -------- ---------- ---------- ---------- ---------- --------
   1    0.3     0.0    0  576543      48      82      13 ( 0.001%)   23321 ( 0.2%) 9928000000000.000 -1.522e+14 -9928000000000.000      0.000      0.000      N/A
   2    0.2     0.5    0  346400      12      29       0 ( 0.000%)   23345 ( 0.2%) 9928000000000.000 -1.632e+14 -9928000000000.000      0.000      0.000      N/A
Restoring best routing
Critical path: 9.928e+12 ns
Successfully routed after 2 routing iterations.
```

litghost, 2019-05-31:
> How accurate/realistic is the global network modelled in this architecture? Are all the RR node co-ordinates correct?

It is accurate in the sense that the graph is attempting to model a power plane and vias from that power plane into the interconnect fabric.  The "bounding box" for a power plane is by definition the entire grid, and the shorted channels attempt to model that.

In generally talking about RR node coordinates for a power plane is kind of nonsensical.  The metal layer is technically speaking everywhere, so what is it's coordinate?

litghost, 2019-05-31:
> Inspecting the RR graph shows that node 934244 connects to 993481 (which connects to many other nodes).
> However 993481 is reported to have location (1,155) <-> (147,155), which falls out side the net's bounding box 31,22 x 50,40 and so is not explored.

But 934244 and 993481 are connected with a short?  I thought the idea was that on shorted nets, the bounding box would either expand to include the entire shorted group?

kmurray, 2019-05-31:
>  I thought the idea was that on shorted nets, the bounding box would either expand to include the entire shorted group?

Nope, the bounding box is not changed by an encounter with a set of shorted nodes.

kmurray, 2019-05-31:
I've changed how the router handles the no path found case to make it more robust in b823094b7.

I believe it means this case should route at the default BB factor. Can you guys confirm whether this fixes the issue?

litghost, 2019-05-31:
> I've changed how the router handles the no path found case to make it more robust in [b823094](https://github.com/verilog-to-routing/vtr-verilog-to-routing/commit/b823094b79688b0f64a32a463b35740cb9b637de).
> 
> I believe it means this case should route at the default BB factor. Can you guys confirm whether this fixes the issue?

This is nice fallback, and I expected it will work.  @acomodi can you try this out?

acomodi, 2019-06-01:
@litghost, @kmurray: I have just tested the no path found case of b823094. Unfortunately, it still fails with in the same way (even though the warning is outputted):
```
	 This is possible on a fringe node based on low Fc_out, N, and certain lengths.
## Loading routing resource graph took 12.81 seconds (max_rss 2899.0 MiB, delta_rss +35.7 MiB)
  RR Graph Nodes: 1538556
  RR Graph Edges: 4738857
Confirming router algorithm: TIMING_DRIVEN.
---- ------ ------- ---- ------- ------- ------- ----------------- --------------- -------- ---------- ---------- ---------- ---------- --------
Iter   Time    pres  BBs    Heap  Re-Rtd  Re-Rtd Overused RR Nodes      Wirelength      CPD       sTNS       sWNS       hTNS       hWNS Est Succ
      (sec)     fac Updt    push    Nets   Conns                                       (ns)       (ns)       (ns)       (ns)       (ns)     Iter
---- ------ ------- ---- ------- ------- ------- ----------------- --------------- -------- ---------- ---------- ---------- ---------- --------
Warning 91300: No routing path for connection to sink_rr 169305, retrying with full device bounding box
Cannot route from SYN-GND.GND[0] (RR node: 113480 type: SOURCE location: (35,26) class: 0) to BLK-TL-SLICEL.BX[0] (RR node: 169305 type: SINK location: (47,35) class: 13) -- no possible path
Failed to route connection from '$false' to '$auto$alumacc.cc:474:replace_alu$7.slice[2].carry4_full' for net '$false' (#10)
Routing failed.
# Routing took 14.22 seconds (max_rss 2899.0 MiB, delta_rss +35.7 MiB)
Circuit is unroutable with a channel width factor of 500.
VPR failed to implement circuit
The entire flow of VPR took 27.10 seconds (max_rss 2899.0 MiB)
```

kmurray, 2019-06-03:
@acomodi Can you point me at the revison you were using?

acomodi, 2019-06-03:
@kmurray, sure, I have used this revision 7eda868b89eb764e272832b8a2acac8166ee74af, from which I have reverted c42d25e4709f5801e6218dc01aed5d2c6bcf4d26 (this commit reverted the non-configurable nodes changes applied ~1 week ago) and cherry-picked your fix https://github.com/verilog-to-routing/vtr-verilog-to-routing/commit/b823094b79688b0f64a32a463b35740cb9b637de

mithro, 2019-06-03:
@acomodi Can you push this branch somewhere and link @kmurray to the branch?

acomodi, 2019-06-03:
Sure, here it is: https://github.com/antmicro/vtr-verilog-to-routing/tree/re-add-route-changes

vtrfmt, 2019-06-05:
Should now be fixed on master (c74fb1fce). I've confirmed that the test-case provided now works.

The issue was we were not clearing the known path costs correctly which prevented the existing route tree from being re-added to the heap. As a result the 2nd attempt to find a route immediately gave up with no path.

Likely #479 is the same problem.

closed 2019-06-05 by vtrfmt

mithro, 2019-06-05:
Who is @vtrfmt?

kmurray, 2019-06-05:
@mithro That was me, accidentally signed into an account I had created (but never used) for the code reformatting.

mithro, 2019-06-05:
@kmurray I guessed that but wanted to make sure :-)

natanvotre, 2019-12-26:
I got the same error, it seems that this error persists.