verilog-to-routing/vtr-verilog-to-routing #618: Build fails in vpr/src/timing/slack_evaluation.cpp when tbb is installed

mithro, 2019-05-29:
#### Current Behaviour

[Current master](https://github.com/verilog-to-routing/vtr-verilog-to-routing/commit/c6e62aaf4f6aa3b2c7fc8135c603611c2df95c92) fails if you have tbb installed with the following error;

```
[ 96%] Building CXX object vpr/CMakeFiles/libvpr.dir/src/timing/slack_evaluation.cpp.o
/home/travis/vtr-build/blifexplorer/blifexplorer_autogen/UVLADIE3JM/qrc_explorerres.cpp: In function ‘int qInitResources_explorerres()’:
/home/travis/vtr-build/blifexplorer/blifexplorer_autogen/UVLADIE3JM/qrc_explorerres.cpp:2165:5: warning: no previous declaration for ‘int qInitResources_explorerres()’ [-Wmissing-declarations]
 int QT_MANGLE_NAMESPACE(qInitResources_explorerres)()
     ^~~~~~~~~~~~~~~~~~~~~~~~~~
/home/travis/vtr-build/blifexplorer/blifexplorer_autogen/UVLADIE3JM/qrc_explorerres.cpp: In function ‘int qCleanupResources_explorerres()’:
/home/travis/vtr-build/blifexplorer/blifexplorer_autogen/UVLADIE3JM/qrc_explorerres.cpp:2174:5: warning: no previous declaration for ‘int qCleanupResources_explorerres()’ [-Wmissing-declarations]
 int QT_MANGLE_NAMESPACE(qCleanupResources_explorerres)()
     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
[ 96%] Linking CXX executable blifexplorer
[ 96%] Built target blifexplorer
[ 96%] Building CXX object vpr/CMakeFiles/libvpr.dir/src/timing/timing_graph_builder.cpp.o
[ 96%] Building CXX object vpr/CMakeFiles/libvpr.dir/src/timing/timing_util.cpp.o
/home/travis/build/mithro/vtr-verilog-to-routing/vpr/src/timing/slack_evaluation.cpp: In member function ‘void SetupSlackCrit::update_slacks(const tatum::SetupTimingAnalyzer&)’:
/home/travis/build/mithro/vtr-verilog-to-routing/vpr/src/timing/slack_evaluation.cpp:50:6: error: no matching function for call to ‘parallel_for_each(vtr::Range<__gnu_cxx::__normal_iterator<const vtr::StrongId<atom_pin_id_tag>*, std::vector<vtr::StrongId<atom_pin_id_tag>, std::allocator<vtr::StrongId<atom_pin_id_tag> > > > >&, SetupSlackCrit::update_slacks(const tatum::SetupTimingAnalyzer&)::<lambda(auto:1)>)’
     });
      ^
In file included from /home/travis/build/mithro/vtr-verilog-to-routing/vpr/src/timing/slack_evaluation.cpp:11:0:
/usr/include/tbb/parallel_for_each.h:60:6: note: candidate: template<class InputIterator, class Function> void tbb::parallel_for_each(InputIterator, InputIterator, const Function&, tbb::task_group_context&)
 void parallel_for_each(InputIterator first, InputIterator last, const Function& f, task_group_context &context) {
      ^~~~~~~~~~~~~~~~~
/usr/include/tbb/parallel_for_each.h:60:6: note:   template argument deduction/substitution failed:
/home/travis/build/mithro/vtr-verilog-to-routing/vpr/src/timing/slack_evaluation.cpp:50:6: note:   deduced conflicting types for parameter ‘InputIterator’ (‘vtr::Range<__gnu_cxx::__normal_iterator<const vtr::StrongId<atom_pin_id_tag>*, std::vector<vtr::StrongId<atom_pin_id_tag>, std::allocator<vtr::StrongId<atom_pin_id_tag> > > > >’ and ‘SetupSlackCrit::update_slacks(const tatum::SetupTimingAnalyzer&)::<lambda(auto:1)>’)
     });
      ^
In file included from /home/travis/build/mithro/vtr-verilog-to-routing/vpr/src/timing/slack_evaluation.cpp:11:0:
/usr/include/tbb/parallel_for_each.h:68:6: note: candidate: template<class InputIterator, class Function> void tbb::parallel_for_each(InputIterator, InputIterator, const Function&)
 void parallel_for_each(InputIterator first, InputIterator last, const Function& f) {
      ^~~~~~~~~~~~~~~~~
/usr/include/tbb/parallel_for_each.h:68:6: note:   template argument deduction/substitution failed:
/home/travis/build/mithro/vtr-verilog-to-routing/vpr/src/timing/slack_evaluation.cpp:50:6: note:   deduced conflicting types for parameter ‘InputIterator’ (‘vtr::Range<__gnu_cxx::__normal_iterator<const vtr::StrongId<atom_pin_id_tag>*, std::vector<vtr::StrongId<atom_pin_id_tag>, std::allocator<vtr::StrongId<atom_pin_id_tag> > > > >’ and ‘SetupSlackCrit::update_slacks(const tatum::SetupTimingAnalyzer&)::<lambda(auto:1)>’)
     });
      ^
/home/travis/build/mithro/vtr-verilog-to-routing/vpr/src/timing/slack_evaluation.cpp: In member function ‘void SetupSlackCrit::update_criticalities(const tatum::TimingGraph&, const tatum::SetupTimingAnalyzer&)’:
/home/travis/build/mithro/vtr-verilog-to-routing/vpr/src/timing/slack_evaluation.cpp:103:6: error: no matching function for call to ‘parallel_for_each(vtr::Range<__gnu_cxx::__normal_iterator<const vtr::StrongId<atom_pin_id_tag>*, std::vector<vtr::StrongId<atom_pin_id_tag>, std::allocator<vtr::StrongId<atom_pin_id_tag> > > > >&, SetupSlackCrit::update_criticalities(const tatum::TimingGraph&, const tatum::SetupTimingAnalyzer&)::<lambda(auto:2)>)’
     });
      ^
In file included from /home/travis/build/mithro/vtr-verilog-to-routing/vpr/src/timing/slack_evaluation.cpp:11:0:
/usr/include/tbb/parallel_for_each.h:60:6: note: candidate: template<class InputIterator, class Function> void tbb::parallel_for_each(InputIterator, InputIterator, const Function&, tbb::task_group_context&)
 void parallel_for_each(InputIterator first, InputIterator last, const Function& f, task_group_context &context) {
      ^~~~~~~~~~~~~~~~~
/usr/include/tbb/parallel_for_each.h:60:6: note:   template argument deduction/substitution failed:
/home/travis/build/mithro/vtr-verilog-to-routing/vpr/src/timing/slack_evaluation.cpp:103:6: note:   deduced conflicting types for parameter ‘InputIterator’ (‘vtr::Range<__gnu_cxx::__normal_iterator<const vtr::StrongId<atom_pin_id_tag>*, std::vector<vtr::StrongId<atom_pin_id_tag>, std::allocator<vtr::StrongId<atom_pin_id_tag> > > > >’ and ‘SetupSlackCrit::update_criticalities(const tatum::TimingGraph&, const tatum::SetupTimingAnalyzer&)::<lambda(auto:2)>’)
     });
      ^
In file included from /home/travis/build/mithro/vtr-verilog-to-routing/vpr/src/timing/slack_evaluation.cpp:11:0:
/usr/include/tbb/parallel_for_each.h:68:6: note: candidate: template<class InputIterator, class Function> void tbb::parallel_for_each(InputIterator, InputIterator, const Function&)
 void parallel_for_each(InputIterator first, InputIterator last, const Function& f) {
      ^~~~~~~~~~~~~~~~~
/usr/include/tbb/parallel_for_each.h:68:6: note:   template argument deduction/substitution failed:
/home/travis/build/mithro/vtr-verilog-to-routing/vpr/src/timing/slack_evaluation.cpp:103:6: note:   deduced conflicting types for parameter ‘InputIterator’ (‘vtr::Range<__gnu_cxx::__normal_iterator<const vtr::StrongId<atom_pin_id_tag>*, std::vector<vtr::StrongId<atom_pin_id_tag>, std::allocator<vtr::StrongId<atom_pin_id_tag> > > > >’ and ‘SetupSlackCrit::update_criticalities(const tatum::TimingGraph&, const tatum::SetupTimingAnalyzer&)::<lambda(auto:2)>’)
     });
      ^
/home/travis/build/mithro/vtr-verilog-to-routing/vpr/src/timing/slack_evaluation.cpp: In member function ‘void HoldSlackCrit::update_criticalities(const tatum::TimingGraph&, const tatum::HoldTimingAnalyzer&)’:
/home/travis/build/mithro/vtr-verilog-to-routing/vpr/src/timing/slack_evaluation.cpp:200:6: error: no matching function for call to ‘parallel_for_each(vtr::Range<__gnu_cxx::__normal_iterator<const vtr::StrongId<atom_pin_id_tag>*, std::vector<vtr::StrongId<atom_pin_id_tag>, std::allocator<vtr::StrongId<atom_pin_id_tag> > > > >&, HoldSlackCrit::update_criticalities(const tatum::TimingGraph&, const tatum::HoldTimingAnalyzer&)::<lambda(auto:3)>)’
     });
      ^
In file included from /home/travis/build/mithro/vtr-verilog-to-routing/vpr/src/timing/slack_evaluation.cpp:11:0:
/usr/include/tbb/parallel_for_each.h:60:6: note: candidate: template<class InputIterator, class Function> void tbb::parallel_for_each(InputIterator, InputIterator, const Function&, tbb::task_group_context&)
 void parallel_for_each(InputIterator first, InputIterator last, const Function& f, task_group_context &context) {
      ^~~~~~~~~~~~~~~~~
/usr/include/tbb/parallel_for_each.h:60:6: note:   template argument deduction/substitution failed:
/home/travis/build/mithro/vtr-verilog-to-routing/vpr/src/timing/slack_evaluation.cpp:200:6: note:   deduced conflicting types for parameter ‘InputIterator’ (‘vtr::Range<__gnu_cxx::__normal_iterator<const vtr::StrongId<atom_pin_id_tag>*, std::vector<vtr::StrongId<atom_pin_id_tag>, std::allocator<vtr::StrongId<atom_pin_id_tag> > > > >’ and ‘HoldSlackCrit::update_criticalities(const tatum::TimingGraph&, const tatum::HoldTimingAnalyzer&)::<lambda(auto:3)>’)
     });
      ^
In file included from /home/travis/build/mithro/vtr-verilog-to-routing/vpr/src/timing/slack_evaluation.cpp:11:0:
[... cut ...] trongId<atom_pin_id_tag>*, std::vector<vtr::StrongId<atom_pin_id_tag>, std::allocator<vtr::StrongId<atom_pin_id_tag> > > > >’ and ‘HoldSlackCrit::update_criticalities(const tatum::TimingGraph&, const tatum::HoldTimingAnalyzer&)::<lambda(auto:3)>’)
     });
      ^
make[3]: *** [vpr/CMakeFiles/libvpr.dir/src/timing/slack_evaluation.cpp.o] Error 1
make[3]: *** Waiting for unfinished jobs....
make[2]: *** [vpr/CMakeFiles/libvpr.dir/all] Error 2
make[1]: *** [all] Error 2
make: *** [all] Error 2
```

kmurray, 2019-05-30:
Which version of TBB are you using? (It seems to work for me with 2017-U7 on Ubuntu 18.04.)

mithro, 2019-05-30:
Build job is at https://travis-ci.org/mithro/vtr-verilog-to-routing/jobs/538988874

mithro, 2019-05-30:
Looks like;
```
Get:114 http://us-east-1.ec2.archive.ubuntu.com/ubuntu trusty/universe amd64 libtbb2 amd64 4.2~20130725-1.1ubuntu1 [94.7 kB]
Get:115 http://us-east-1.ec2.archive.ubuntu.com/ubuntu trusty/universe amd64 libtbb-dev amd64 4.2~20130725-1.1ubuntu1 [176 kB]
```

kmurray, 2019-05-31:
Should be fixed in 9b2aaf695.

Seems like older versions of TBB don't support passing a container/range to tbb::parallel_for_each, and some compilers are more picky about capturing `this` in lambdas.

closed 2019-05-31 by kmurray