verilog-to-routing/vtr-verilog-to-routing #620: Port, Test & Tune RCV++ algorithm for hold time resolution to master

vaughnbetz, 2019-05-30:
#### Proposed Behaviour
We should have a new --router_algorithm <rcv> (or pick your favourite name) that turns on hold time resolution in the VPR router.

#### Current Behaviour
The router optimizes only for setup time.

#### Possible Solution
Ming has a branch with an enhanced RCV algorithm that appears to resolve hold quite well, with reasonable wire use and CPU time. The data structures updates needed by this algorithm should be ported to the master, after checking they don't slow down the current (timing_driven) router or bloat memory footprint. Then this should become a new, optional router after sufficient testing and tuning. The QoR data should be attached as spreadsheets here so we can track it.  We'll need two sets of data: how the current timing_driven router is affected by the data structure updates (hopefully not much) and how well rcv works.

#### Context
Designs with complex clocking often have hold issues. If you have significant clock skew, you can also have hold issues even with simple clocking (no inter-clock synchronous transfers).  So we need an algorithm that can fix these. In addition, there is some evidence the rcv algorithm can improve setup time as well, so we may also obtain a circuit speed up with this change.

This algorithm will not be a complete fix, as hold time violations for paths completely contained within a cluster can't be fixed by the inter-cluster router. Fixing the within-cluster violations will be a separate follow-up issue after this one is finished.

@renrichard

litghost, 2019-08-08:
Is there a working version of the RCV++ algo that can be tried?  I'm seeing some hold closure.  I also wonder if the placer needs some knowledge or idea of hold violations. 

Anyways, if there is a WIP branch with RCV working, I'd gladly check it out on the designs I'm seeing issues with.

vaughnbetz, 2019-08-08:
It's mostly fixed in routing, rather than placement.  RCV is being debugged by Ming and Richard; I wouldn't try it yet but hopefully soon.
@renrichard @evozone96: feel free to give an update here.

renrichard, 2019-08-10:
@vaughnbetz @litghost 

@evozone96  and I just discussed our plans forward.
1. I am running all the regression tests with RCV turned off to ensure they pass.
2. We are currently debugging LU32PEEng from the VTR benchmarks with RCV turned on.
3. Once we resolve this, @evozone96 will push a branch onto the master.

kmurray, 2019-08-13:
@renrichard @evozone96  I think getting the RCV work onto a branch in this repo is an important first step. That will ensure both of you can work collaboratively on debugging.

After that:
1. We need to get all existing regression tests working (no errors/crashes)
2. Verify is that regular (with RCV turned off) routing quality and run-time are no worse than the current implementation.

Once that is done, even if RCV isn't fully working yet, we could consider merging the code into master, with RCV off by default. Work could then continue on master to finish debugging RCV.

closed 2025-05-15 by github-actions[bot]