verilog-to-routing/vtr-verilog-to-routing #621: Packer fails to pack all the possible nets in a single pb_type

acomodi, 2019-05-30:
#### Expected Behaviour

The packer packs all the nets in only one top level tile (named `TILE`), instead of using two of them, causing a `not enough resources` error.

#### Current Behaviour

In the architecture setting present in the archive there is a small test case that fails to correctly finish VPR successfully as it uses more tiles than the one present on the device. This because (as visible in the `.net` file), the `ADDER` primitive, which has multiple pb_type instances, is underused for each `TILE`, even though there is enough "space" to fit all the nets in only one `TILE`. (see also the `.eblif`)

#### Possible Solution

I have tested all the possible VPR command line options that could solve the problem:
- `--allow_unrelated_clustering on`
- `--target_ext_pin_util 1.0,1.0`
- `--enable_pin_feasibility_filter off` (and removing `--target_ext_pin_util`)
Unfortunately, I had no luck. I am pretty sure the issue is in the architecture XML definition, but I have double-checked it and it seems to be fine.

#### Steps to Reproduce

The VTR version is irrelevant, it happens with many revisions, but for correctness I have checked it with both 567188a0c and 40f04d9bba1d4a607eee38a4f9305e702af3b0b2 with the same result.

[multiple_instance.zip](https://github.com/verilog-to-routing/vtr-verilog-to-routing/files/3237470/multiple_instance.zip)
1. Extract the .zip 
2. `cd multiple_instance`
3. Run: 
```
vpr multiple_instance.arch.merged.xml multiple_instance.test.eblif --echo_file on --device device --target_ext_pin_util 1.0,1.0 --disp on --pack_verbosity 4 --allow_unrelated_clustering on --pack --place --route
```

#### Context

VPR clustering step. There is the need to create a series of tests for SymbiFlow which use a auto-generated architecture and `.eblif` for single `pb_types`.

#### Your Environment

* VTR revision used: 
* Operating System and version: Linux/Ubuntu 18.04
* Compiler version:

vaughnbetz, 2019-06-20:
@acomodi I think you're saying this circuit should fit in one tile, not two, but the "Expected Behaviour" lists the opposite.  Does that need to be corrected?

Generally packing as densely as possible isn't a very high priority, as in large circuits a very dense packing usually causes more routability problems than its worth (there are exceptions, but this still makes the priority lower).

acomodi, 2019-06-20:
@vaughnbetz As you correctly said the circuit should fit in only one tile, not two.
To be more precise:
- The architecture is made of a number of IO tiles which connect to a single "central" tile. 
- The circuit is built so that all the pins and all the tiles (IO and the "central" one) are being used.
- The "central" tile contains a number of primitive pb_types enough to cover all the nets in the circuit.

What happens is that the "central" tile is only partly used by the cluster and another one is needed, making the packer fail as there are not enough "central" tiles on the device.

acomodi, 2020-12-08:
Closing. Fixed in https://github.com/verilog-to-routing/vtr-verilog-to-routing/pull/926

closed 2020-12-08 by acomodi