verilog-to-routing/vtr-verilog-to-routing #622: Fix Odin-II memory leaks

vaughnbetz, 2019-05-30:
#### Expected Behaviour
vpr and odin-II should have no valgrind errors or warnings.  When there are some warnings (the current case) it is hard to see new memory leaks or errors, so it makes it harder to maintain code quality.

#### Current Behaviour
When run through the valgrind regtest, Odin-II has memory leaks.  vpr is memory clean. The perl interpreter used to run the regression flow also has some small memory leaks, but those are not our concern.

#### Possible Solution
1. Fix Odin-II memory leaks.
2. Change the regtest to not run valgrind on the perl interpreter (lower priority).

#### Steps to Reproduce
Results are at: http://builds.verilogtorouting.org:8080/builders/strong/builds/864/steps/vtr_reg_valgrind_small/logs/stdio
http://builds.verilogtorouting.org:8080/builders/nightly/builds/774/steps/vtr_reg_valgrind/logs/stdio

The 2 regtests that produce these results are encoded in the link name.

#### Your Environment
VTR trunk, build is in the links above.

jeanlego, 2019-07-16:
@j-b-1-7

jeanlego, 2019-07-22:
@vaughnbetz as of recently, vtr_reg_valgrind_small should run with no leaks from Odin.
would there be any way to silence the Perl warnings to be able to catch only the warnings we do care about? I think silencing them would allow us to catch if any leaks are reintroduced

kmurray, 2019-07-22:
> would there be any way to silence the Perl warnings
@jeanlego I think the way to do this is to update all the calls to perl scripts to not be wrapped by valgrind

jeanlego, 2019-09-18:
@sdamghan can you confirm that basic_vtr contains no leak?

sdamghan, 2019-09-18:
@jeanlego NO! it still has mem leaks, I've checked it a few mins before.

jeanlego, 2019-09-18:
please report ETA here also so we can track how close we are to being leak free

sdamghan, 2019-09-20:
Basic_vtr contains no memory leak issues. It has been checked through the Valgrind for all tasks. This issue can be closed.

vaughnbetz, 2019-09-20:
Yeah!  Thanks.

closed 2019-09-20 by vaughnbetz