verilog-to-routing/vtr-verilog-to-routing #623: Odin-II fails with code sanitizers enabled

vaughnbetz, 2019-05-30:
#### Expected Behaviour
Sanitizers should show no errors on the basic and strong regression tests.

#### Current Behaviour
Odin-II fails the sanitizers; they flag issues and then exit.
If run on its own, vpr should be sanitizer clean (it was at last check, but since we can't check automatically there is a chance of regression).

#### Possible Solution
Fix the bugs flagged by the sanitizers.

#### Steps to Reproduce
See developer documents for building with sanitizers; then run the basic and strong tests.

#### Context
Prevents automatic reg test runs with sanitizers on, which risks code quality. The bugs flagged by the sanitizers may also indicate issues that compromise Odin-II stability.

#### Your Environment
Was vtr master; last checked 6 months ago.

vaughnbetz, 2019-09-19:
Is this fixed now?

jeanlego, 2019-09-23:
@sdamghan or @msohrabi2019  can you have a look?

jeanlego, 2019-09-24:
@vaughnbetz, this has been fixed and should be good to close!

closed 2019-09-24 by vaughnbetz