verilog-to-routing/vtr-verilog-to-routing #624: Support include_what_you_use running on CI

mithro, 2019-05-30:
#### Proposed Behaviour

[`include-what-you-use` is a tool](https://include-what-you-use.org/) which makes sure that your C and C++ files include the correct set of headers. This makes it easier to refactor code as it doesn't stop working randomly because a include is missing.

#### Current Behaviour

You can accidentally forget to add an include but the code still works because something else includes it.

#### Context

This is another code quality improvement tool.

kmurray, 2019-05-31:
I agree cleaning up extra includes etc is good.

Any thoughts on just doing this periodically vs running on CI? 

I know CI is going to be more consistent, but this does seem like a lower priority issue.

mithro, 2019-06-02:
@kmurray Once it has been cleaned up, I think we should just run it as part of the C++ formatting check job.

vaughnbetz, 2019-06-20:
This makes sense to me.  Tim, do you want to drive it?  Happy to assign it to you, and put it in the C++ formatting check.

vaughnbetz, 2025-05-15:
This is a good idea ... @AmirhosseinPoolad discuss and see if you can take this one on.

AlexandreSinger, 2025-05-15:
I also agree that this is a good idea. I have found that several files in VTR have header files that they include which they do not use. This is most problematic in header files. Some header files include other header files that they do not use which causes a cascade of includes. This ultimately leads to a problem when modifying a single header file ends up causing practically all files in VTR to be rebuilt which is not ideal.

Transitive includes (including a header file that you do not use, but you use something that it includes) also has similar problems but are not nearly as bad, especially if we have already cleaned up truly unused header files.

I have this tool installed locally in my editor and have found that many files in VTR have includes which are unused. I have tried removing them when I can, but they are very tricky to remove due to header files including unused header files which many of the CPP files rely on.

PR #3048 (2025-05-17): Add documentation for include sanitization
Adds documentation explaining how you can use IWYU or clangd diagnostics to clean up includes. This is the first step in trying to slowly fix #624. The number of include errors we currently have is very high and I don't think it's feasible to fix the entire thing in one go, so I'm proposing to start with making anyone that touches a file in a PR responsible for fixing the includes for that file (This should probably and ideally be enforced by CI). Doing this should slowly make the code cleaner and maybe in the future we can add an IWYU stage to CI.

N [... cut]
closed 2025-05-17 by AlexandreSinger