verilog-to-routing/vtr-verilog-to-routing #625: Explicitly document which versions of GCC / clang are supported

mithro, 2019-05-31:
#### Proposed Behaviour

It would be good if the documentation explicitly said which versions of GCC and clang were supported. We can then run make sure these versions compile on the CI system.

FYI - The compilers released by Ubuntu can be found here;
https://gist.github.com/mithro/85c97dd594a8726d081c57b2b541630e

#### Possible Solution

I suggest the following list;
 - [ ] gcc-5 -- To support default compiler on Ubuntu Xenial (which is an LTS).
 - [ ] gcc-6 -- To support default compiler on Debian Stretch.
 - [ ] gcc-8 -- To support Debian testing + unstable and Ubuntu bionic (LTS) + cosmic.
 - [ ] gcc-9 -- To support Ubuntu disco + eoan.
 - [ ] clang-3.9 -- Earliest reasonable version of clang.
 - [ ] clang-8 -- Latest released version of clang.

I don't use Redhat / Fedora, so I don't know if this covers a good range of compilers.

mithro, 2019-05-31:
It might be worth supporting `clang-6` which seems to be available in Debian testing+unstable plus Ubuntu xenial -> eoan.

mithro, 2019-05-31:
Here is an example of running the compiles after Build / Testing is a success
 * Commit -> https://github.com/verilog-to-routing/vtr-verilog-to-routing/commit/438a97566b96a5e2342ef8e7eee98f9d59fd2e05
 * Example build -> https://travis-ci.org/mithro/vtr-verilog-to-routing/builds/539482119

mithro, 2019-05-31:
Looks like the build bot checks the following configurations;
https://github.com/verilog-to-routing/vtr-verilog-to-routing/blob/81d1673afb080246b7269cd23230b3007fbec654/sweep_build_configs.py#L14-L16

mithro, 2019-05-31:
@kmurray -- This is a question for you and @vaughnbetz

mithro, 2019-05-31:
Just FYI the current status seems to be;

 - GCC 5 works
 - GCC 6 works
 - GCC 8 fails due to LTO issue #628 
 - GCC 9 fails due to LTO issue #628 
 - clang 3.6 fails with #629 
 - clang 6.0 currently running...
 - clang 8 fails due to an LTO issue (mayve #628)

kmurray, 2019-05-31:
What is in sweep_build_configs.py is what we currently test the code builds against. And we generally try to keep it warning clean for the GCC compilers.

Effectively our requirement is a C++14 compliant compiler.

However I've been contemplating dropping gcc-4.9, since:
* Ubuntu 14.04 LTS is now EOL
* It isn't fully C++14 compliant (there are a couple corners of the syntax which we've had to work around). For example see [this build error](http://builds.verilogtorouting.org:8080/builders/basic/builds/935/steps/make/logs/stdio).

I'm leaning towards making the VTR 8.0 release with gcc-4.9 tested, and then drop it from the list of tested compilers shortly afterwards.

mithro, 2019-06-02:
@kmurray It might be better to drop gcc-4.9 support *before* VtR 8.0 release to stop people using it as an excuse to use VtR 8.0 verses upstream.

mithro, 2019-06-04:
Update on the status! Thanks to @kmurray's fixing of #628 we are pretty close to having everything working. See https://travis-ci.org/verilog-to-routing/vtr-verilog-to-routing/builds/541097847

Just FYI the current status seems to be;
 - GCC 5 works
 - GCC 6 works
 - GCC 8 works (seemed to have fixed #628)
 - GCC 9 works (seemed to have issue #628)
 - clang 3.6 fails with #629 
 - clang 6 works
 - clang 8 works

So, it only looks like clang-3.6 is broken!

litghost, 2019-06-12:
FYI, I'm seeing test failures on GCC 6 and clang 7:
```
soft_multipliers:         k6_frac_N10_4add_2chains_depop50_mem20K_22nm.xml/mult_5x5.v/common                             failed: vpr (exited with return code 2) (took 3.02 seconds)
multiclock:               k6_frac_N10_mem32K_40nm.xml/multiclock.blif/common                                             OK              (took 1.40 seconds)
soft_multipliers:         k6_frac_N10_4add_2chains_depop50_mem20K_22nm.xml/mult_8x8.v/common                             OK              (took 4.94 seconds)
soft_multipliers:         k6_frac_N10_4add_2chains_depop50_mem20K_22nm.xml/mult_6x6.v/common                             OK              (took 5.07 seconds)
no_timing:                k6_frac_N10_frac_chain_depop50_mem32K_40nm.xml/ch_intrinsics.v/common                          failed: vpr (exited with return code 139) (took 5.14 seconds)
eblif_vpr:                k6_frac_N10_40nm.xml/test_eblif.eblif/common                                                   OK              (took 1.15 seconds)
fracturable_luts:         k6_N8_I80_fleI10_fleO2_ff2_nmodes_2.xml/ch_intrinsics.v/common                                 failed: vpr (exited with return code 139) (took 5.73 seconds)

```

vaughnbetz, 2019-06-20:
Kevin can't reproduce these either; do they still occur?

mithro, 2019-09-10:
@acomodi Can you close out this issue (IE Check that everything is the same and then close it.)

acomodi, 2019-09-11:
@mithro I have checked that all the current builds (GCC-5/6/7/8/9 and Clang-6/8) do pass the strong regression tests. I can confirm this can be closed now.
I am unable to close the issue though, probably because I wasn't the original author of the issue.

closed 2019-09-11 by mithro

vaughnbetz, 2019-09-11:
Thanks @acomodi!