verilog-to-routing/vtr-verilog-to-routing #628: Internal compiler error when building with GCC 8.0 (on Travis)

mithro, 2019-05-31:
#### Current Behaviour

See build here -> https://travis-ci.org/mithro/vtr-verilog-to-routing/jobs/539520885

```
lto1: internal compiler error: in add_symbol_to_partition_1, at lto/lto-partition.c:136
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-8/README.Bugs> for instructions.
lto-wrapper: fatal error: /usr/bin/g++-8 returned 1 exit status
compilation terminated.
/usr/bin/ld: lto-wrapper failed
collect2: error: ld returned 1 exit status
make[3]: *** [vpr/test_vpr] Error 1
make[2]: *** [vpr/CMakeFiles/test_vpr.dir/all] Error 2
make[2]: *** Waiting for unfinished jobs....
```

mithro, 2019-05-31:
This [also seems to affect GCC 8.0](https://travis-ci.org/mithro/vtr-verilog-to-routing/jobs/539520886).

I'm wondering if there is an issue with the ancient version of `/usr/bin/ld` on Travis Trusty and the output from more modern compilers?

mithro, 2019-05-31:
[clang-8 also seems to have a similar type of error.](https://travis-ci.org/mithro/vtr-verilog-to-routing/jobs/539520889)
```
libvpr.a: error adding symbols: Archive has no index; run ranlib to add one
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[3]: *** [vpr/test_vpr] Error 1
make[2]: *** [vpr/CMakeFiles/test_vpr.dir/all] Error 2
make[2]: *** Waiting for unfinished jobs....
/usr/bin/ld: /usr/lib/llvm-8/bin/../lib/LLVMgold.so: error loading plugin: /usr/lib/llvm-8/bin/../lib/LLVMgold.so: cannot open shared object file: No such file or directory
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[3]: *** [vpr/vpr] Error 1
make[2]: *** [vpr/CMakeFiles/vpr.dir/all] Error 2
```

jeanlego, 2019-05-31:
See issue #530.

mithro, 2019-05-31:
@jeanlego We should figure out why vtr causes lto to be unhappy.

kmurray, 2019-05-31:
Yep I've seen this on occasion as well. It seems to be somewhat sporadic, but since it appears to be a gcc error I haven't looked into more detail.

If someone wants to dig into it go for it :)

kmurray, 2019-06-03:
[This](https://github.com/Project-OSRM/osrm-backend/issues/2951) hints it may be an issue with inconsistent compilation flags.

I've re-worked (203c3b532) how we were handling LTO so that both VPR and all the libraries it depends on (e.g. libvtrutil, libarchfpga, libtatum) are built with the same flags (including LTO). That seems to have fixed the case which I was encountering.

As a side effect the various other tools (odin, ace) are now also built with LTO (instead of just VPR). The only exception is ABC were I've explicitly left LTO off since its an externally developed tool.

We now also require cmake >= 3.9 which has full support for LTO.

mithro, 2019-06-04:
This seems to have been fixed!

https://travis-ci.org/verilog-to-routing/vtr-verilog-to-routing/jobs/541097858

mithro, 2019-06-04:
```
--
Using linker
GNU ld (GNU Binutils for Ubuntu) 2.24
Copyright 2013 Free Software Foundation, Inc.
This program is free software; you may redistribute it under the terms of
the GNU General Public License version 3 or (at your option) a later version.
This program has absolutely no warranty.
GNU gold (GNU Binutils for Ubuntu 2.24) 1.11
Copyright 2013 Free Software Foundation, Inc.
This program is free software; you may redistribute it under the terms of
the GNU General Public License version 3 or (at your option) a later version.
This program has absolutely no warranty.
--
Using compiler C compiler 'gcc-8'
gcc-8 (Ubuntu 8.1.0-5ubuntu1~14.04) 8.1.0
Copyright (C) 2018 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
--
Using compiler C++ compiler 'g++-8'
g++-8 (Ubuntu 8.1.0-5ubuntu1~14.04) 8.1.0
Copyright (C) 2018 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

closed 2019-06-04 by mithro