verilog-to-routing/vtr-verilog-to-routing #629: VtR fails to compile with clang-3.6

mithro, 2019-05-31:
#### Current Behaviour

https://travis-ci.org/mithro/vtr-verilog-to-routing/jobs/539544488

```
--
Using compiler C compiler 'clang-3.6'
Ubuntu clang version 3.6.2-svn240577-1~exp1 (branches/release_36) (based on LLVM 3.6.2)
Target: x86_64-pc-linux-gnu
Thread model: posix
--
Using compiler C++ compiler 'clang++-3.6'
Ubuntu clang version 3.6.2-svn240577-1~exp1 (branches/release_36) (based on LLVM 3.6.2)
Target: x86_64-pc-linux-gnu
Thread model: posix
--
```

```
/home/travis/build/mithro/vtr-verilog-to-routing/libs/libvtrutil/src/vtr_ndmatrix.h:38:26: warning: comparison of unsigned expression >= 0 is always true [-Wtautological-compare]
        do { if (!(index >= 0)) { vtr::assert::handle_assert("index >= 0", "/home/travis/build/mithro/vtr-verilog-to-routing/libs/libvtrutil/src/vtr_ndmatrix.h", 38, __func__, "Index out of range (below dimension minimum)"); } } while (false);
                   ~~~~~ ^  ~
/home/travis/build/mithro/vtr-verilog-to-routing/libs/libvtrutil/src/vtr_ndmatrix.h:78:26: warning: comparison of unsigned expression >= 0 is always true [-Wtautological-compare]
        do { if (!(index >= 0)) { vtr::assert::handle_assert("index >= 0", "/home/travis/build/mithro/vtr-verilog-to-routing/libs/libvtrutil/src/vtr_ndmatrix.h", 78, __func__, "Index out of range (below dimension minimum)"); } } while (false);
                   ~~~~~ ^  ~
/home/travis/build/mithro/vtr-verilog-to-routing/libs/libvtrutil/src/vtr_ndmatrix.h:305:26: warning: comparison of unsigned expression >= 0 is always true [-Wtautological-compare]
        do { if (!(index >= 0)) { vtr::assert::handle_assert("index >= 0", "/home/travis/build/mithro/vtr-verilog-to-routing/libs/libvtrutil/src/vtr_ndmatrix.h", 305, __func__, "Index out of range (below dimension minimum)"); } } while (false);
                   ~~~~~ ^  ~
/home/travis/build/mithro/vtr-verilog-to-routing/libs/libvtrutil/src/vtr_ndmatrix.h:341:26: warning: comparison of unsigned expression >= 0 is always true [-Wtautological-compare]
        do { if (!(index >= 0)) { vtr::assert::handle_assert("index >= 0", "/home/travis/build/mithro/vtr-verilog-to-routing/libs/libvtrutil/src/vtr_ndmatrix.h", 341, __func__, "Index out of range (below dimension minimum)"); } } while (false);
                   ~~~~~ ^  ~
/home/travis/build/mithro/vtr-verilog-to-routing/vpr/src/route/route_tree_timing.cpp:1308:28: error: no matching member function for call to 'erase'
    route_tree_connections.erase(std::make_tuple(OPEN, OPEN, rt_root->inode));
    ~~~~~~~~~~~~~~~~~~~~~~~^~~~~
/usr/bin/../lib/gcc/x86_64-linux-gnu/6.5.0/../../../../include/c++/6.5.0/bits/stl_set.h:571:7: note: candidate function not viable: no known conversion from 'tuple<typename __decay_and_strip<e_pin_type>::__type, typename __decay_and_strip<e_pin_type>::__type, typename __decay_and_strip<const int &>::__type>' to 'const_iterator' (aka '_Rb_tree_const_iterator<value_type>') for 1st argument
      erase(const_iterator __position)
      ^
/usr/bin/../lib/gcc/x86_64-linux-gnu/6.5.0/../../../../include/c++/6.5.0/bits/stl_set.h:601:7: note: candidate function not viable: no known conversion from 'tuple<typename __decay_and_strip<e_pin_type>::__type, typename __decay_and_strip<e_pin_type>::__type, [...]>' to 'const tuple<int, int, [...]>' for 1st argument
      erase(const key_type& __x)
      ^
/usr/bin/../lib/gcc/x86_64-linux-gnu/6.5.0/../../../../include/c++/6.5.0/bits/stl_set.h:623:7: note: candidate function not viable: requires 2 arguments, but 1 was provided
      erase(const_iterator __first, const_iterator __last)
      ^
4 warnings and 1 error generated.
make[3]: *** [vpr/CMakeFiles/libvpr.dir/src/route/route_tree_timing.cpp.o] Error 1
make[2]: *** [vpr/CMakeFiles/libvpr.dir/all] Error 2
make[1]: *** [all] Error 2
make: *** [all] Error 2
The command "./.github/travis/build.sh" exited with 2.
```

mithro, 2019-06-04:
https://api.travis-ci.org/v3/job/541097860/log.txt
```
In file included from /home/travis/build/verilog-to-routing/vtr-verilog-to-routing/vpr/src/route/route_tree_timing.cpp:11:
In file included from /home/travis/build/verilog-to-routing/vtr-verilog-to-routing/vpr/src/base/vpr_types.h:30:
In file included from /home/travis/build/verilog-to-routing/vtr-verilog-to-routing/libs/libarchfpga/src/arch_types.h:12:
In file included from /home/travis/build/verilog-to-routing/vtr-verilog-to-routing/libs/libarchfpga/src/physical_types.h:38:
/home/travis/build/verilog-to-routing/vtr-verilog-to-routing/libs/libvtrutil/src/vtr_ndmatrix.h:38:26: warning: comparison of unsigned expression >= 0 is always true [-Wtautological-compare]
        do { if (!(index >= 0)) { vtr::assert::handle_assert("index >= 0", "/home/travis/build/verilog-to-routing/vtr-verilog-to-routing/libs/libvtrutil/src/vtr_ndmatrix.h", 38, __func__, "Index out of range (below dimension minimum)"); } } while (false);
                   ~~~~~ ^  ~
/home/travis/build/verilog-to-routing/vtr-verilog-to-routing/libs/libvtrutil/src/vtr_ndmatrix.h:78:26: warning: comparison of unsigned expression >= 0 is always true [-Wtautological-compare]
        do { if (!(index >= 0)) { vtr::assert::handle_assert("index >= 0", "/home/travis/build/verilog-to-routing/vtr-verilog-to-routing/libs/libvtrutil/src/vtr_ndmatrix.h", 78, __func__, "Index out of range (below dimension minimum)"); } } while (false);
                   ~~~~~ ^  ~
/home/travis/build/verilog-to-routing/vtr-verilog-to-routing/libs/libvtrutil/src/vtr_ndmatrix.h:305:26: warning: comparison of unsigned expression >= 0 is always true [-Wtautological-compare]
        do { if (!(index >= 0)) { vtr::assert::handle_assert("index >= 0", "/home/travis/build/verilog-to-routing/vtr-verilog-to-routing/libs/libvtrutil/src/vtr_ndmatrix.h", 305, __func__, "Index out of range (below dimension minimum)"); } } while (false);
                   ~~~~~ ^  ~
/home/travis/build/verilog-to-routing/vtr-verilog-to-routing/libs/libvtrutil/src/vtr_ndmatrix.h:341:26: warning: comparison of unsigned expression >= 0 is always true [-Wtautological-compare]
        do { if (!(index >= 0)) { vtr::assert::handle_assert("index >= 0", "/home/travis/build/verilog-to-routing/vtr-verilog-to-routing/libs/libvtrutil/src/vtr_ndmatrix.h", 341, __func__, "Index out of range (below dimension minimum)"); } } while (false);
                   ~~~~~ ^  ~
/home/travis/build/verilog-to-routing/vtr-verilog-to-routing/vpr/src/route/route_tree_timing.cpp:1305:28: error: no matching member function for call to 'erase'
    route_tree_connections.erase(std::make_tuple(OPEN, OPEN, rt_root->inode));
    ~~~~~~~~~~~~~~~~~~~~~~~^~~~~
/usr/bin/../lib/gcc/x86_64-linux-gnu/6.5.0/../../../../include/c++/6.5.0/bits/stl_set.h:571:7: note: candidate function not viable: no known conversion from 'tuple<typename __decay_and_strip<e_pin_type>::__type, typename __decay_and_strip<e_pin_type>::__type, typename __decay_and_strip<const int &>::__type>' to 'const_iterator' (aka '_Rb_tree_const_iterator<value_type>') for 1st argument
      erase(const_iterator __position)
      ^
/usr/bin/../lib/gcc/x86_64-linux-gnu/6.5.0/../../../../include/c++/6.5.0/bits/stl_set.h:601:7: note: candidate function not viable: no known conversion from 'tuple<typename __decay_and_strip<e_pin_type>::__type, typename __decay_and_strip<e_pin_type>::__type, [...]>' to 'const tuple<int, int, [...]>' for 1st argument
      erase(const key_type& __x)
      ^
/usr/bin/../lib/gcc/x86_64-linux-gnu/6.5.0/../../../../include/c++/6.5.0/bits/stl_set.h:623:7: note: candidate function not viable: requires 2 arguments, but 1 was provided
      erase(const_iterator __first, const_iterator __last)
      ^
4 warnings and 1 error generated.
make[3]: *** [vpr/CMakeFiles/libvpr.dir/src/route/route_tree_timing.cpp.o] Error 1
make[2]: *** [vpr/CMakeFiles/libvpr.dir/all] Error 2
make[1]: *** [all] Error 2
make: *** [all] Error 2
travis_time:end:0a11798e:start=1559635530757096910,finish=1559636013649654661,duration=482892557751
[0K[31;1mThe command "./.github/travis/build.sh" exited with 2.[0m
```

vaughnbetz, 2019-06-20:
Kevin tells me this compiles OK for him (didn't reproduce). Is this still happening?  We could get Richard to recode the offending line if necessary, but it's not clear why it doesn't match, nor why it doesn't reproduce for Kevin.

closed 2025-05-15 by github-actions[bot]