verilog-to-routing/vtr-verilog-to-routing #635: Odin II Syntax Error on OpenCL examples

suyang5, 2019-06-05:
While running OpenCL examples through Odin II (worked in Quartus II), a variety of syntax errors occured.
Link for OpenCL examples: [OpenCL examples](https://www.intel.com/content/www/us/en/programmable/products/design-software/embedded-software-developers/opencl/support.html#supportedplatforms)
Quartus compatible version of 2 examples: 
[fft1d_kernel_odinII.tar.gz](https://github.com/verilog-to-routing/vtr-verilog-to-routing/files/3258521/fft1d_kernel_odinII.tar.gz)
[mandelbrot_realKernel_odinII.tar.gz](https://github.com/verilog-to-routing/vtr-verilog-to-routing/files/3258523/mandelbrot_realKernel_odinII.tar.gz)

#### Expected Behaviour

It is expected to convert the entire verilog project into a .blif format file

#### Current Behaviour

Syntax error occured. 
The most representative one is `(syntax error, unexpected vINTEGRAL, expecting ';' or ',' or '.')::97`. 
Another common one is `(syntax error, unexpected '#', expecting '(')::-1`.

#### Possible Solution

#### Steps to Reproduce

1. Extract any of the compressed files uploaded
2. Change the directory in terminal to the extracted folder
3. enter command `./odin_II -c odinConfig.xml`
4. The syntax error will show up in the terminal

Note: One example (compressed file) represent one of the two common syntax errors encountered.

#### Context

I am trying to push the OpenCL example through the VTR flow

#### Your Environment

* VTR revision used: Odin_II 0.1
* Operating System and version: Ubuntu 14.04.5 LTS

jeanlego, 2019-06-08:
It seems there are a lot of VHD and SV files in the list. 
Odin does not handle other languages than Verilog for the moment. 

Working on improving the error message in these cases.

thanks for the detailed report!

jeanlego, 2019-06-22:
Closing as now we report proper errors when we encounter SV and vhdl files

closed 2019-06-22 by jeanlego