verilog-to-routing/vtr-verilog-to-routing #640: Odin fails when executing run_vtr_flow.pl

a181d129, 2019-06-06:
I am trying to run run_vtr_flow.pl script.After going to the vtr_flow/tasks directory, I wrote the following command :
../scripts/run_vtr_task.pl basic_flow
It gives me the following message :
k6_frac_N10_mem32K_40nm.xml/ch_intrinsics.v/common 
                           failed: odin
				I am confused about the reason of failure. could someone help me out?
I am running the vtr-verilog-to-routing-master tool from windows machine using WinSCP

jeanlego, 2019-06-08:
this seems related to your setup

could you put the output of the file ? 

> vtr_flow/tasks/basic_flow/run001/k6_frac_N10_mem32K_40nm.xml/ch_intrinsics.v/common/odin.out

jeanlego, 2019-06-22:
Unable to reproduce, closing

closed 2019-06-22 by jeanlego