verilog-to-routing/vtr-verilog-to-routing #653: Timing lower bound analysis for flow

litghost, 2019-06-17:
#### Proposed Behaviour

As part of debugging https://github.com/verilog-to-routing/vtr-verilog-to-routing/issues/592 and https://github.com/SymbiFlow/symbiflow-arch-defs/issues/716 , a question was raised by @HackerFoo about adding analysis to tease apart where in the flow the timing margin is being taken.  I am proposing the following analysis's to provide lower bounds on critical path delay (CPD).  The core idea is to identify where in the flow margin is being consumed.  

The idea here is to provide guidance on where the timing margin is being consumed. For example, if the lower bound CPD after pack shows negative or zero margin to the constraints, then design changes or synthesis and packing improvements would be required to recover timing margin.

1. Post-synthesis timing analysis
 - Perform STA using the minimum max delay path for matching blackboxes.  This will ignore routing delays and pb_type connection delays, as this analysis is intended to prior to packing.  The critical path in this analysis is expected to underestimate the CPD significantly.  However, the intention of this analysis is simply to provide a lower bound on CPD.
2. Post-packing timing analysis
 - I can foresee two analysises possible here.
    1.  The first is a straight forward zero routing delay analysis.  Again this will dramatically under predicate the delay, but provide a strong lower bound for the CPD. 
    2. The second analysis could be a congestion-free delay estimate.  For nets that connect from the OPIN of a cluster to the IPIN of the same cluster, an optimal route delay from OPIN to IPIN can be computed (and cached) using any tile of that cluster type.  For nets that connect from one cluster to another cluster, an optimal route delay from IPIN class to OPIN class could be devised via a Dijkstra search that excluded self-connections.  The current RR node structure does not allow to easy reverse graph walking, other heuristic measures could be taken.
3. Post-placement timing analysis
 - A congestion-free Dijkstra routing delay estimate could be found, which would provide a precise lower bound for the design post-placement.

#### Current Behaviour

I need to investigate the current analysis behavior and determine to what degree the existing analyses provide lower bounds.

#### Context

Currently VPR is failing to close timing on a handful of design using the XC7 routing graph.  It is unclear at this time whether the problem lies with packing, placement, or routing.  There was a problem in the synthesis that was identified and resolved, so the question is now where to focus the attention next.

The main idea is by providing precise lower bounds where possible, it becomes possible to systematically attack the problem, when/if an obvious violation occurs.  For example, if the precise post-packing analysis shows a very small or negative CPD margin, then that clearly points at an issue upstream of place/route flows.

I believe that it is likely not a good idea to excessively optimize the earlier part of the flow based on these lower bound analyses, but having some idea of the lower bound can guide future work and debugging.

litghost, 2019-06-17:
@kmurray @vaughnbetz Thoughts?  I'm planning to focus on this analysis this week to see if I can tease out where in the flow to target.  I do need to survey how the existing post-placement timing analysis works.  I don't believe VPR has a post-synthesis analysis at all, but I could be mistaken.

My next step is to review and document what the current flow analyses are.

litghost, 2019-06-17:
It looks like something akin to what I propose for the post-synthesis analysis exists, but no report is generated, specifically the https://github.com/verilog-to-routing/vtr-verilog-to-routing/blob/master/vpr/src/timing/PreClusterDelayCalculator.h is used for criticality analysis for timing based packing.  However a timing report is not being generated using this setup analysis.  There is also the different that the PreClusterDelayCalculator is attempting to estimate inter cluster delay, which prevents the analysis from being a true lower bound.

I'll see if I can add an echo output which is the pre-cluster delay setup report.  This report will be useful in debugging if the packer is tracking the correct critical path, and provide the beginning of a lower bound report.

vaughnbetz, 2019-06-17:
3. pretty much exists right now; probably just needs some high level code to call for a report and a command line option.  

I suggest dividing 3 into two timing reports. You could use the placer estimated timing (estimates without congestion) and/or the post-first-iteration routing timing analysis.  Both those estimates can be useful to debug the placement delay estimator and to see if routing congestion is impacting your critical path.  If you don't get a full timing report now for those phases I think you could add optional code to generate a timing report at each phase of the tool, and control it with a new command line option.

1. and 2. are also doable as you suggest.  They're usually less useful, as they are not very tight bounds. Usually when optimizing a design (in VTR or Quartus) I instead look at what fraction of the critical path comes from various things (logic vs. routing, and how far the routing is going) to decide what to optimize.  But if you think they are useful I don't think they would take too much code, and you could add them to that "analyze after each phase" command line option.  The assume 0 routing delay version in particular isn't very hard to get going.

If you want a slightly tighter bound that considers lower bound routing delays:
You shouldn't need to run a Djikstra search backward from IPINs to OPINs to find the minimum possible delay path from any placement location to another location of the proper type.  You can run a forward Djikstra from each OPIN type to all interesting IPIN types.  This is somewhat related to what the placer delay estimate builder does, but you only care about nearby blocks (and the fastest of all of them in the end).  You may be able to get this information from the placement delay estimator right now, as its per dx/dy distance info is a superset of what you'd need. Kevin added some detailed options to find and cache delays to / from different OPINs and IPINs, but he's the best person to ask for the details of the state of that code (which I think is off by default).

litghost, 2019-06-17:
> 1. and 2. are also doable as you suggest.  They're usually less useful, as they are not very tight bounds.

I just generated the pre-clustering timing report, and I think I've identified some bugs in the way pre-cluster timing analysis is being done.  The bug is two-fold:

- Inter-cluster delay is being used for direct connections, which resulting in dramatically overestimating the path delay.  Example:
```
$auto$maccmap.cc:240:synth$4026.slice[0].carry4_1st_full.CO_CHAIN[0] (CARRY4_VPR)                       0.528     9.855
$auto$maccmap.cc:240:synth$4026.slice[1].carry4_full.CIN[0] (CARRY4_VPR)                                2.180    12.035
$auto$maccmap.cc:240:synth$4026.slice[1].carry4_full.CO_CHAIN[0] (CARRY4_VPR)                           0.114    12.149
$auto$maccmap.cc:240:synth$4026.slice[2].carry4_full.CIN[0] (CARRY4_VPR)                                2.180    14.329
$auto$maccmap.cc:240:synth$4026.slice[2].carry4_full.CO_CHAIN[0] (CARRY4_VPR)                           0.114    14.443
```
The delay between CO_CHAIN -> CIN should be the direct cost (rather than the intercluster delay estimate).  In this case the direct cost is 0 ns.  So the path delay is being over-estimated by 7*2.180 ~= 15.26 ns.

- Intra-cluster net delays are always using the inter-cluster delay estimate.  This is causing dramatic over-estimates of the delay for cluster paths.  It is less clear how to handle this, but take the following example:
```
murax.system_cpu.RegFilePlugin_regFile.0.0.1.DPRA[0].in[2] (.names)                                      2.180    27.344
murax.system_cpu.RegFilePlugin_regFile.0.0.1.DPRA[0].out[0] (.names)                                     0.150    27.494
murax.system_cpu.RegFilePlugin_regFile.18.0.1.dram0.A[0] (DPRAM64)                                       2.180    29.674
murax.system_cpu.RegFilePlugin_regFile.18.0.1.dram0.O[0] (DPRAM64)                                       0.124    29.798
murax.system_cpu.RegFilePlugin_regFile.18.0.1.stub.DPO[0] (DRAM_2_OUTPUT_STUB)                           2.180    31.978
murax.system_cpu.RegFilePlugin_regFile.18.0.1.stub.DPO_OUT[0] (DRAM_2_OUTPUT_STUB)                       0.000    31.978
$auto$simplemap.cc:420:simplemap_dff$8508.D[0] (FDRE_ZINI)                                               2.180    34.158
```

DPRAM64 and DRAM_2_OUTPUT_STUB will be joined in pre-packing, so the delay between them should be determined from the pre-pack clustering.  In fact, DRAM_2_OUTPUT_STUB cannot actually connect to the routing fabric at all.  It is an open question whether the FF (FDRE) would actually cluster with the LUT-RAM output, but in general it probably will be done.

I think maybe a first order pruning of inter-cluster might be worthwhile?  For example:
 - If two atoms will be joined in pre-packing, do not add the intercluster delay between them.  If we are getting fancy, could find the cost from all possible cluster placements?
 - If two atoms use a chain pattern (e.g. Fc = 0), use the cost on the direct connection rather than inter cluster delay
 - If atoms can not be connected within the cluster, always use the inter cluster delay.  Example being a LUT followed by a LUT would always have the inter cluster delay
 - Given that most CLB/SLICE's have registration (e.g. LUT -> FF), maybe assign a fractional registration cost, rather than the full inter-cluster delay?

It is unclear to me the degree in which the pre-cluster delay (used for criticality) matters for packing, but from this look, it seems the pre-cluster delay is especially poor.

vaughnbetz, 2019-06-17:
Thanks for the analysis. Fixing the chain pattern would be worthwhile, in my opinion, as we'll be way off in timing estimates without that fix.  That will skew where we think the critical path is a lot. So the first two changes make sense to me.

LUTs and FFs are usually pre-packed into molecules as well, so I think if you pre-packing to molecules to estimate delays correctly you'll fix carry chains and LUT->FF simultaneously.

A LUT to LUT connection can often be made via a faster intra-cluster routing (in most architectures).  So I wouldn't estimate that as always inter-cluster for all architectures.  You could estimate connections that cannot be intra cluster (e.g. LUT -> mult, or LUT->LUT in architectures with no local interconnect) to always be inter-cluster. It would be best to key that off the architecture so we avoid all sorts of hard-coding (work out what the lower-bound delay is for various primitive -> primitive connections).  It doesn't have to be a perfect estimate, but again would ideally adapt to the architecture.

The pre-clustering delay estimate can't be made extremely accurate, and in general doesn't have to be very accurate as the clusterer tries to minimize all sorts of possibly critical connections and capture lots of nets to improve routability.  But gross errors like not flagging carry chains may mess quality up some.  Basically the earlier you are in the CAD flow the more timing uncertainty there is in what happens downstream, and hence the more the early algorithms focus on optimizing many paths / capturing many nets rather than focusing particularly on a smaller set of timing-critical connections.

kmurray, 2019-06-17:
>     (1) Post-synthesis timing analysis

I'd call this a pre-pack timing analysis. To get what you'd want I think you just need to set the [estimated inter-cluster delay](https://github.com/verilog-to-routing/vtr-verilog-to-routing/blob/master/vpr/src/base/vpr_api.cpp#L496-L498) to zero.

> (2.i) Post-pack zero routing delay analysis

You'd just need to [zero out the point_to_point_delay or net_delay arrays before passing them to the post-cluster delay calculator](https://github.com/verilog-to-routing/vtr-verilog-to-routing/blob/master/vpr/src/place/place.cpp#L539)

>  (3) pretty much exists right now; probably just needs some high level code to call for a report and a command line option.

There are already options for this, see:
  * Post-placement: [--post_place_timing_report](https://docs.verilogtorouting.org/en/latest/vpr/command_line_usage/#cmdoption-vpr-post-place-timing-report)
  * Congestion-oblivious routed delay: [--router_first_iter_timing_report](https://docs.verilogtorouting.org/en/latest/vpr/command_line_usage/#cmdoption-vpr-router-first-iter-timing-report)

litghost, 2019-06-17:
> I'd call this a pre-pack timing analysis. To get what you'd want I think you just need to set the [estimated inter-cluster [...]

Thanks. I figured this might exist.

litghost, 2019-06-18:
> * Congestion-oblivious routed delay: [--router_first_iter_timing_report](https://docs.verilogtorouting.org/en/latest/vpr/command_line_usage/#cmdoption-vpr-router-first-iter-timing-report)

So I've found something very strange when I used this option is that the router choose not to use the dedicated clock network on the first iteration.  This behavior surprises me greatly, as trying to use the general interconnect to route the clock results in a very poor delay value:

```
io_mainClk.inpad[0] (.input at (42,26))                                                                                        0.000     0.000
| (intra 'SYN-INPAD' routing)                                                                                                  0.010     0.010
| (inter-block routing)                                                                                                       19.137    19.147
| (intra 'BLK-TL-SLICEL' routing)                                                                                              0.000    19.147
```

By iteration 2, this behavior seems to be gone.  Tomorrow I'll dig into why this might be the case, but I'm wondering if the router starting with such a poor clock routing could permanently degrade the solution?  Or because of the initial poor route will the router potentially fail to rip up routes and stay in a "bad" initial space?

vaughnbetz, 2019-06-18:
@mustafabbas Adding Mustafa to comment on the clock routing.  Hopefully this will be fixed when the two-phase clock routing algorithm is put back, which should be soon.

litghost, 2019-06-18:
> @mustafabbas Adding Mustafa to comment on the clock routing. Hopefully this will be fixed when the two-phase clock routing algorithm is put back, which should be soon.

It is worth noting we have not annotated our clock networks in the rr graph yet, I don't know if that has been documented.  Is there special node/edge annotations for dedicated clock networks?

vaughnbetz, 2019-06-18:
I am pretty sure it is not documented yet.  Also on the to-do list.  @mustafabbas: @litghost is sending in an rr-graph through the read an entire rr-graph flow, so we'll need to document what flag/tag data must be set on the clock root or any other special nodes. Basically the rr-graph reader should support setting the same tags/flags that the rr-graph creation code does for global clocks.

closed 2025-05-15 by github-actions[bot]