verilog-to-routing/vtr-verilog-to-routing #655: CI failure

acomodi, 2019-06-18:
#### Expected Behaviour

CI correctly builds

#### Current Behaviour

CI is failing due to the fact that all the various tests are unable to find the executable files.
An example of log failure is the following:
```
Could not find executable /home/travis/vtr-build/libs/libvtrutil/test_vtrutil
Looked in the following places:
/home/travis/vtr-build/libs/libvtrutil/test_vtrutil
/home/travis/vtr-build/libs/libvtrutil/test_vtrutil
/home/travis/vtr-build/libs/libvtrutil/Release/test_vtrutil
/home/travis/vtr-build/libs/libvtrutil/Release/test_vtrutil
/home/travis/vtr-build/libs/libvtrutil/Debug/test_vtrutil
/home/travis/vtr-build/libs/libvtrutil/Debug/test_vtrutil
/home/travis/vtr-build/libs/libvtrutil/MinSizeRel/test_vtrutil
/home/travis/vtr-build/libs/libvtrutil/MinSizeRel/test_vtrutil
/home/travis/vtr-build/libs/libvtrutil/RelWithDebInfo/test_vtrutil
/home/travis/vtr-build/libs/libvtrutil/RelWithDebInfo/test_vtrutil
/home/travis/vtr-build/libs/libvtrutil/Deployment/test_vtrutil
/home/travis/vtr-build/libs/libvtrutil/Deployment/test_vtrutil
/home/travis/vtr-build/libs/libvtrutil/Development/test_vtrutil
/home/travis/vtr-build/libs/libvtrutil/Development/test_vtrutil
home/travis/vtr-build/libs/libvtrutil/test_vtrutil
home/travis/vtr-build/libs/libvtrutil/test_vtrutil
home/travis/vtr-build/libs/libvtrutil/Release/test_vtrutil
home/travis/vtr-build/libs/libvtrutil/Release/test_vtrutil
home/travis/vtr-build/libs/libvtrutil/Debug/test_vtrutil
home/travis/vtr-build/libs/libvtrutil/Debug/test_vtrutil
home/travis/vtr-build/libs/libvtrutil/MinSizeRel/test_vtrutil
home/travis/vtr-build/libs/libvtrutil/MinSizeRel/test_vtrutil
home/travis/vtr-build/libs/libvtrutil/RelWithDebInfo/test_vtrutil
home/travis/vtr-build/libs/libvtrutil/RelWithDebInfo/test_vtrutil
home/travis/vtr-build/libs/libvtrutil/Deployment/test_vtrutil
home/travis/vtr-build/libs/libvtrutil/Deployment/test_vtrutil
home/travis/vtr-build/libs/libvtrutil/Development/test_vtrutil
home/travis/vtr-build/libs/libvtrutil/Development/test_vtrutil
Unable to find executable: /home/travis/vtr-build/libs/libvtrutil/test_vtrutil
1/3 Test #1: test_vtrutil .....................***Not Run   0.00 sec
    Start 2: test_vpr
...
```
Strangely, the issue showed up with https://github.com/verilog-to-routing/vtr-verilog-to-routing/commit/d016fffe50ab77f059dbea7c972cbf70a37f1fa8, which does not change anything in the testing infrastructure.

#### Context

VtR CI

kmurray, 2019-06-18:
Yep I've noticed this too.

I expect something has changed on the Travis side probably related to caching.
I've looked at the logs and the key difference is:

[Working:](https://travis-ci.org/verilog-to-routing/vtr-verilog-to-routing/builds/546774489)
```
Setting up build cache
$ export CASHER_DIR=${TRAVIS_HOME}/.casher
0.15s$ Installing caching utilities
0.00s3.40sattempting to download cache archive
fetching master/cache-linux-trusty-e3b0c44298fc1c149afbf4c8996fb92427ae41e4649b934ca495991b7852b855--compiler-gpp.tgz
found cache
0.00s8.35sadding /home/travis/vtr-build to cache
creating directory /home/travis/vtr-build
adding /home/travis/vtr to cache
creating directory /home/travis/vtr
cache.ccache
0.00s8.51sadding /home/travis/.ccache to cache
$ g++ --version
g++ (Ubuntu 4.8.4-2ubuntu1~14.04.3) 4.8.4
```

[Broken:](https://travis-ci.org/verilog-to-routing/vtr-verilog-to-routing/jobs/546800363)
```
Setting up build cache
$ export CASHER_DIR=${TRAVIS_HOME}/.casher
0.16s$ Installing caching utilities
0.00s3.28sattempting to download cache archive
fetching master/cache-linux-trusty-e3b0c44298fc1c149afbf4c8996fb92427ae41e4649b934ca495991b7852b855--compiler-gpp.tgz
found cache
0.00s7.75sadding /home/travis/vtr-build to cache
creating directory /home/travis/vtr-build
adding /home/travis/vtr to cache
creating directory /home/travis/vtr
/home/travis/vtr-build is not yet cached
/home/travis/vtr is not yet cached
cache.ccache
0.00s8.70sadding /home/travis/.ccache to cache
$ g++ --version
g++ (Ubuntu 4.8.4-2ubuntu1~14.04.3) 4.8.4
```

Where for some reason the vtr and vtr-build dirs are not being cached correctly.

@mithro I think you setup the caching system, any ideas?

kmurray, 2019-06-18:
It appears that Travis CI is now passing (with no code changes, indicating this was a Travis issue).

closed 2019-06-18 by kmurray