verilog-to-routing/vtr-verilog-to-routing #658: Variable "effort" based on criticality

litghost, 2019-06-19:
#### Proposed Behaviour/Solution

bb factor and A* variable vary based on slack.  If the nets along the path meet timing, the just use default bb factor and A*.  Each iteration that a net is along the path does not meet timing, and the router chooses to "rip up", the bb factor and A* factor change to increase the effort the router searches for a route.  The idea here is that critical path nets deserve more effort, rather than increasing the effort uniformly.

#### Current Behaviour

bb factor and A* are constant across nets.

#### Context

I'm currently exploring making VPR produce "good" solutions on 7-series graph architecture, and have found that increase bb factor (effectively removing the bounding box) and lowering the A* factor (1.2 -> 0, 0.4), have been important for closing timing.

I'm currently working on making a demo using the Murax SoC close at a clock of 100 MHz.  Vivado took the same synthesis and was "only" able to reach a slack of 1.434 ns (e.g. critical path delay of 8.566 ns).

Here is the current results I've been getting so far (with various A* and bb factor values):

A* factor | bb factor | Runtime (sec) | CPD (ns) | Freq (MHz)
-- | -- | -- | -- | --
1.6 | Default | 762.04 | 15.1719 | 65.9113
0 | 10000 |   |   |  
1.4 | 10000 | 2629.47 | 14.2745 | 70.0552
0.4 | 10 |   |   |  
1 | 100 | 3147.56 | 11.1741 | 89.4929
1 | 50 |   |   |  
1 | 25 | 3037.93 | 15.0482 | 66.453
0.6 | 10 | 3470.07 | 11.0593 | 90.4215
0.6 | 20 |   |   |  
0.6 | 100 |   |   |  
1.2 | 100 |   |   |  

The rows in the table that lack values are still running.  The A* factor = 0 / bb factor = 10000 locks like it is approaching a VPR estimated CPD of 8.517 ns, which is a very reasonable result, and closes timing.  However the runtime for for this setting is very prohibitive.  It has currently been running for 4 hrs.

litghost, 2019-06-19:
Example routing logs

A* factor = 0, bb factor = 10000:
```
---- ------ ------- ---- ------- ------- ------- ----------------- --------------- -------- ---------- ---------- ---------- ---------- --------
Iter   Time    pres  BBs    Heap  Re-Rtd  Re-Rtd Overused RR Nodes      Wirelength      CPD       sTNS       sWNS       hTNS       hWNS Est Succ
      (sec)     fac Updt    push    Nets   Conns                                       (ns)       (ns)       (ns)       (ns)       (ns)     Iter
---- ------ ------- ---- ------- ------- ------- ----------------- --------------- -------- ---------- ---------- ---------- ---------- --------
Warning 271068: 17 timing startpoints were not constrained during timing analysis
Warning 271069: 18 timing endpoints were not constrained during timing analysis
   1 6379.7     0.0    0 1.5e+10    2661    8648    4495 ( 0.415%)  222599 ( 5.4%)   19.604     -404.7     -9.604     -145.0    -12.912      N/A
   2 1006.9     0.5    0 3.5e+09    2099    6550    3858 ( 0.356%)  158068 ( 3.8%)    8.780      0.000      0.000     -32.69     -3.321      N/A
   3 1366.9     0.6    0 4.4e+09    2008    6112    3276 ( 0.302%)  160241 ( 3.9%)    8.521      0.000      0.000      0.000      0.000      N/A
   4  906.6     0.8    0 2.8e+09    1906    5760    3389 ( 0.313%)  159127 ( 3.9%)    8.517      0.000      0.000      0.000      0.000      N/A
   5 1003.2     1.1    0 2.7e+09    1833    5475    2737 ( 0.253%)  159883 ( 3.9%)    8.517      0.000      0.000      0.000      0.000      N/A
   6  882.6     1.4    0 2.3e+09    1672    4991    2204 ( 0.203%)  160852 ( 3.9%)    8.517      0.000      0.000      0.000      0.000      N/A
   7  744.7     1.9    0 2.0e+09    1472    4398    1695 ( 0.156%)  160975 ( 3.9%)    8.517      0.000      0.000      0.000      0.000      N/A
   8  657.4     2.4    0 1.7e+09    1321    3866    1368 ( 0.126%)  162065 ( 3.9%)    8.517      0.000      0.000      0.000      0.000      N/A
   9  551.0     3.1    0 1.5e+09    1156    3399    1030 ( 0.095%)  162674 ( 3.9%)    8.517      0.000      0.000      0.000      0.000      N/A
  10  488.2     4.1    0 1.2e+09     963    2816     768 ( 0.071%)  163623 ( 4.0%)    8.517      0.000      0.000      0.000      0.000       38
  11  429.2     5.3    0 1.1e+09     787    2347     553 ( 0.051%)  164564 ( 4.0%)    8.517      0.000      0.000      0.000      0.000       36
  12  362.7     6.9    0 8.9e+08     611    1840     360 ( 0.033%)  166068 ( 4.0%)    8.517      0.000      0.000      0.000      0.000       34
  13  265.5     9.0    0 6.4e+08     411    1199     201 ( 0.019%)  167344 ( 4.1%)    8.517      0.000      0.000      0.000      0.000       31
  14  215.2    11.6    0 5.0e+08     267     755     103 ( 0.010%)  168199 ( 4.1%)    8.517      0.000      0.000      0.000      0.000       29
  15  137.4    15.1    0 3.0e+08     152     415      52 ( 0.005%)  168953 ( 4.1%)    8.897      0.000      0.000      0.000      0.000       26
  16   76.4    19.7    0 1.6e+08      75     227      29 ( 0.003%)  169434 ( 4.1%)    8.873      0.000      0.000      0.000      0.000       24
  17   47.2    25.6    0 9.8e+07      44     122      10 ( 0.001%)  169532 ( 4.1%)    8.873      0.000      0.000      0.000      0.000       23
  18   24.9    33.3    0 5.0e+07      23      56       6 ( 0.001%)  169781 ( 4.1%)    8.873      0.000      0.000      0.000      0.000       22
  19   15.2    43.3    0 3.0e+07      13      36       2 ( 0.000%)  169837 ( 4.1%)    8.873      0.000      0.000      0.000      0.000       21
  20   10.3    56.2    0 1.9e+07       8      17       2 ( 0.000%)  169846 ( 4.1%)    8.873      0.000      0.000      0.000      0.000       21
  21   11.7    73.1    0 2.2e+07       9      18       2 ( 0.000%)  169827 ( 4.1%)    8.873      0.000      0.000      0.000      0.000       21
  22    8.4    95.0    0 1.6e+07       7      17       2 ( 0.000%)  169837 ( 4.1%)    8.873      0.000      0.000      0.000      0.000       21
  23    9.7   123.5    0 1.9e+07       8      17       2 ( 0.000%)  169865 ( 4.1%)    8.873      0.000      0.000      0.000      0.000       22
  24    8.8   160.6    0 1.8e+07       8      17       1 ( 0.000%)  169864 ( 4.1%)    8.873      0.000      0.000      0.000      0.000       22
  25    5.0   208.8    0 1.0e+07       5      11       1 ( 0.000%)  169875 ( 4.1%)    8.873      0.000      0.000      0.000      0.000       23
  26    8.1   271.4    0 1.6e+07       7      14       1 ( 0.000%)  169893 ( 4.1%)    8.873      0.000      0.000      0.000      0.000       23
  27    6.5   352.8    0 1.3e+07       6      12       1 ( 0.000%)  169875 ( 4.1%)    8.873      0.000      0.000      0.000      0.000       24
  28    5.1   458.7    0 1.0e+07       5      11       1 ( 0.000%)  169886 ( 4.1%)    8.873      0.000      0.000      0.000      0.000       25
  29    6.7   596.3    0 1.3e+07       6      12       1 ( 0.000%)  169902 ( 4.1%)    8.873      0.000      0.000      0.000      0.000       25
  30    5.8   775.1    0 1.1e+07       6      12       1 ( 0.000%)  169920 ( 4.1%)    8.873      0.000      0.000      0.000      0.000       26
  31    4.2  1007.7    0 8544744       5      11       1 ( 0.000%)  169902 ( 4.1%)    8.873      0.000      0.000      0.000      0.000       27
  32    7.5  1310.0    0 1.4e+07       7      13       1 ( 0.000%)  169889 ( 4.1%)    8.873      0.000      0.000      0.000      0.000       27
[... cut ...]
  26    1.2   271.4    0 2826408       2       2       1 ( 0.000%)  173611 ( 4.2%)   11.174     -14.74     -1.174     -348.8     -4.904       25
  27    1.2   352.8    0 2803552       2       2       1 ( 0.000%)  173621 ( 4.2%)   11.174     -14.74     -1.174     -348.8     -4.904       25
  28    1.3   458.7    0 3833159       3       9       2 ( 0.000%)  173604 ( 4.2%)   11.174     -14.74     -1.174     -347.7     -4.904       25
  29    0.2   596.3    0 1033242       2       4       0 ( 0.000%)  173442 ( 4.2%)   11.174     -15.01     -1.174     -345.5     -4.904       26
```

vaughnbetz, 2019-06-20:
Thanks for the data; it's interesting.  I think the data shows there are still major lookahead issues though, as astar_fac = 1 is not performing well.  So a per-net (or per-connection) astar could be added, but before doing that and trying to tune it I think we need to get the lookahead in a reasonable state.

bb_factor is normally a moderate runtime optimization (once you have a good lookahead and are running with astar_fac somewhere near 1). With astar_fac near 0, smaller bb_factors are much more important to control runtime, but it'll still be bad even with small bb_factor.

So I think it's too soon to start adding fine-tuning knobs; there are deeper lookahead issues that should be fixed first. Also, once you get into fine-tuning algorithms you'll need to use a larger benchmark suite as you only want to do debugging and very coarse tuning on a single circuit.

litghost, 2019-06-20:
> Thanks for the data; it's interesting. I think the data shows there are still major lookahead issues though, as astar_fac = 1 is not performing well. So a per-net (or per-connection) astar could be [...]

Thanks for the feedback.  I'll gather some more data on lookahead behavior.  It is worth noting that this new lookahead is doing better, but if you think based on this data that the lookahead is still not good enough, I'll focus my energy there.

The good outcome here is that I have a placement that is "good enough" to close timing, I just need to enable the router to reach it.  This is important in my eyes because it means I can ignore problems in packing and placement and focus on routing behavior.   

To be clear, there may be additional gains to be had, but the pack and place is good enough (by existance proof) to focus on the router behavior.

vaughnbetz, 2019-06-20:
Yes, that is good news.  In an uncongested design, I'd expect the router to quickly converge to a solution with near best-case routing delay, with astar somewhere around 1.  Until that happens, modifying the fundamental algorithm is probably not a good idea, as you're working around a bad lookahead and more likely to detune than tune the algorithm for the case when the lookahead works well.