verilog-to-routing/vtr-verilog-to-routing #724: Coverity Scan Errors in ODIN II

vaughnbetz, 2019-07-15:
#224 ## Current Behaviour
There are two ODIN-II pieces of code flagged by the latest coverity scan.  They should be fixed to be coverity clean.  They are:

________________________________________________________________________________________________________
*** CID 201657:  Resource leaks  (RESOURCE_LEAK)
/home/aaron/workspace/vtr/vtrrepo/ODIN_II/SRC/parse_making_ast.cpp: 1430 in newFunctionInstance(char *, ast_node_t *, int)()
1424     
1425            /* store the module symbol name that this calls in a list that will at the end be asociated with the module node */
1426            function_instantiations_instance_by_module = (ast_node_t **)vtr::realloc(function_instantiations_instance_by_module, sizeof(ast_node_t*)*(size_function_instantiations_by_module+1));
1427            function_instantiations_instance_by_module[size_function_instantiations_by_module] = new_node;
1428            size_function_instantiations_by_module++;
1429     
>>>     CID 201657:  Resource leaks  (RESOURCE_LEAK)
>>>     Variable "symbol_node" going out of scope leaks the storage it points to.
1430            return new_node;
1431     }
1432     
1433     /*---------------------------------------------------------------------------------------------
1434      * (function: newGateInstance)
1435      *-------------------------------------------------------------------------------------------*/

________________________________________________________________________________________________________
*** CID 201645:  API usage errors  (PRINTF_ARGS)
/home/aaron/workspace/vtr/vtrrepo/ODIN_II/SRC/simulate_blif.cpp: 3284 in assign_memory_from_mif_file(nnode_t *, _IO_FILE *, char *, int, long)()
3278                                                    item_in = symbols.find("WIDTH");
3279                                                    if ( item_in == symbols.end() ) 
3280                                                            error_message(SIMULATION_ERROR, -1, -1, "%s: MIF WIDTH parameter unspecified.", filename);
3281     
3282                                                    long mif_width = std::strtol(item_in->second.c_str(),NULL,10);
3283                                                    if (mif_width != width)
>>>     CID 201645:  API usage errors  (PRINTF_ARGS)
>>>     Argument "width" to format specifier "%ld" was expected to have type "long" but has type "int".
3284                                                            error_message(SIMULATION_ERROR, -1, -1, "%s: MIF width mismatch: must be %ld but %ld was given", filename, width, mif_width);
3285     
3286     
3287                                                    // Verify the depth parameter.
3288                                                    item_in = symbols.find("DEPTH");
3289                                                    if ( item_in == symbols.end() ) 

#### Your Environment
Coverity scan details can be accessed through the link below.

To view the defects in Coverity Scan visit, https://u2389337.ct.sendgrid.net/wf/click?upn=08onrYu34A-2BWcWUl-2F-2BfV0V05UPxvVjWch-2Bd2MGckcRbAHCEZ8DHqT-2FrK1UDr7mvIT337BBJeoPVU9LcuZ07oPimOMPFsAFFXF3UiO-2BL-2Fm2o-3D_yZ-2FW0SaYYRc8Uzk4ssaRTK-2FXIH6IYaq2CFktAeJeg0OAhMjz4B-2BVcfqXl0aLp39uFhK0WKiRs9tuXHrcNo02akY0FIsaRndEOaL1gFLfNYc09jgpdhKUGkxSwwQlfvCBNAnt595o6u-2BuBGtIkR9Y82qrsT5okAwzASbnCDx1vEh1Lf-2FYX2mQYqHdnvBwKwUY2ucKdIc4ewlHN9KMei1E7WZqgzZQ7kREKGBhHsTHXHo-3D

jeanlego, 2019-07-16:
Fixed

jeanlego, 2019-07-20:
Fixed

closed 2019-07-20 by jeanlego