verilog-to-routing/vtr-verilog-to-routing #732: Unchecked dynamic cast

kmurray, 2019-07-16:
As reported by coverity, there appears to be an unchecked dynamic cast in the clock construction code:
* https://scan9.coverity.com/reports.htm#v27175/p13131/fileInstanceId=52643001&defectInstanceId=6589586&mergedDefectId=201661
* https://scan9.coverity.com/reports.htm#v27175/p13131/fileInstanceId=52643001&defectInstanceId=6588764&mergedDefectId=201909

@mustafabbas I believe this is your code, can you take a look?

kmurray, 2019-07-16:
Probably related:
* https://scan9.coverity.com/reports.htm#v27175/p13131/fileInstanceId=52643001&defectInstanceId=6588238&mergedDefectId=201307

mustafabbas, 2019-11-17:
Hi Kevin, how do I get access to coverity? Or can you past the scan result. Thanks

kmurray, 2019-11-18:
I've sent the invitation, once you register you should be able to see the results.

closed 2025-05-16 by github-actions[bot]