verilog-to-routing/vtr-verilog-to-routing #733: Weekly coverity scan

jeanlego, 2019-07-16:
#### Proposed Behaviour
Have the (weekly?) buildbot generate a Coverity scan.

#### Context
Coverity scan could not be done due to insufficient coverage. It would be beneficial to have Coverity run weekly to have updated reports and help improve the codebase.

jeanlego, 2019-07-24:
@kmurray should we have Travis generate it? Maybe on a commit with a tag or have buildbot generate it from an iirc command?

kmurray, 2019-07-24:
I agree submitting a build to coverity on a regular basis is a good idea!

However, there are some limits related to how often we can submit builds: https://scan.coverity.com/faq

Submitting it once a week seems like a good frequency. I'm not sure how to do a once-a-week submission from Travis, so it probably makes sense to trigger it off of buildbot which already has a scheduled weekly run.

jeanlego, 2019-07-24:
I was thinking something like https://docs.travis-ci.com/user/cron-jobs/

It sets a flag that we could catch and then trigger a different build.
Wether we integrate Coverity scan builds into cmake (or the Makefile) was the next question.

It could be uncoupled and the build step written directly written into to Travis scripts

kmurray, 2019-07-24:
@jeanlego I'd be tempted to keep it decoupled. 

If I recall we just need to install the coverity build tool, and specify it as the compiler to generate all info coverity needs. It then just needs to be uploaded to their server.

jeanlego, 2019-07-26:
I've worked out something, just need to enable the cron job, tested locally it works, don't know about travis much if anyone has better experience with it to chime in

PR #864 (2019-08-03): Travis: include Coverity build step

#### Description
This PR include build step to make and push Coverity builds 
Currently, use my email (should be regenerated by one of the long-term maintainers?)

#### Related Issue
closes #733

#### Motivation and Context
Increase frequency of Coverity scan to improve code quality and lessen errors

#### How Has This Been Tested?

used the travis linter to verify no bug were introduced (any other to test before pushing?)
```travis lint .travis.yml```

tested cron_build.sh localy with token from coverity

#### Types of changes

- [ ] Bug fix (change which  [... cut]
closed 2019-08-03 by jeanlego