verilog-to-routing/vtr-verilog-to-routing #845: Memory Leaks produced by route_tree_timing.cpp

tinayang3024, 2019-07-23:
#### Expected Behaviour
valgrind output (red ones are produced by route_tree_timing.cpp)
https://docs.google.com/document/d/1ldtNS-Pw7e3Lx-FUGzPDOHW5HxBAJgkVTax72PVAop8/edit?usp=sharing

#### Steps to Reproduce

ran valgrind on the head of libezgl
command line used
`valgrind --leak-check=full --errors-for-leak-kinds=none --error-exitcode=1 --track-origins=yes /nfs/ug/thesis/thesis0/vb2019/yangdin1/vtr/vtr-verilog-to-routing/vpr/vpr /nfs/ug/thesis/thesis0/vb2019/yangdin1/vtr/vtr-verilog-to-routing/vtr_flow/arch/timing/k6_frac_N10_frac_chain_mem32K_htree0_40nm.xml /nfs/ug/thesis/thesis0/vb2019/yangdin1/vtr/vtr-verilog-to-routing/vtr_flow/benchmarks/blif/alu4.blif --disp on`

renrichard, 2019-07-23:
On master branch, ran with command: valgrind --leak-check=full --show-leak-kinds=all --errors-for-leak-kinds=none --error-exitcode=1 --track-origins=yes ./cmd.sh

"Only can upload txt version"

[cmd.txt](https://github.com/verilog-to-routing/vtr-verilog-to-routing/files/3423776/cmd.txt)
[output.txt](https://github.com/verilog-to-routing/vtr-verilog-to-routing/files/3423750/output.txt)

kmurray, 2019-07-23:
Looking at the logs provided by @tinayang3024 an example like:
```
==14253== 296 (40 direct, 256 indirect) bytes in 1 blocks are definitely lost in loss record 9,255 of 10,032
==14253==    at 0x4C2BBAF: malloc (vg_replace_malloc.c:299)
==14253==    by 0x423ACE: vtr::malloc(unsigned long) (vtr_memory.cpp:56)
==14253==    by 0x278A3E: alloc_rt_node (route_tree_timing.cpp:145)
==14253==    by 0x278A3E: traceback_to_route_tree_branch(t_trace*, std::map<int, t_rt_node*, std::less<int>, std::allocator<std::pair<int const, t_rt_node*> > >&) (route_tree_timing.cpp:738)
==14253==    by 0x27DFA0: traceback_to_route_tree_branch (route_tree_timing.cpp:763)
==14253==    by 0x27DFA0: traceback_to_route_tree(t_trace*) (route_tree_timing.cpp:713)
==14253==    by 0x1FF277: trace_routed_connection_rr_nodes (draw.cpp:3236)
==14253==    by 0x1FF277: draw_routed_timing_edge_connection (draw.cpp:3211)
==14253==    by 0x1FF277: draw_routed_timing_edge (draw.cpp:3152)
==14253==    by 0x1FF277: draw_crit_path(ezgl::renderer&) (draw.cpp:3100)
==14253==    by 0x200E92: draw_main_canvas(ezgl::renderer&) (draw.cpp:277)
==14253==    by 0x456623: ezgl::canvas::redraw() (canvas.cpp:166)
==14253==    by 0x6BAB1A3: ??? (in /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0.5000.3)
==14253==    by 0x6BC58CC: g_signal_emit_valist (in /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0.5000.3)
==14253==    by 0x6BC5FBE: g_signal_emit (in /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0.5000.3)
==14253==    by 0x4F64D7C: ??? (in /usr/lib/x86_64-linux-gnu/libgtk-3.so.0.2200.11)
==14253==    by 0x4F64DE4: ??? (in /usr/lib/x86_64-linux-gnu/libgtk-3.so.0.2200.11)
```

Seems to show that the leak originates from the drawing code via `draw_crit_path`. It looks like the function `trace_routed_connection_rr_nodes()` calls `traceback_to_route_tree()` (which allocates the route tree). My guess is that this leak was in the code base before the ezgl changes.

@renrichard What is the function to free an allocated route tree? That is probably what we need to call in `trace_routed_connection_rr_nodes()` to ensure the allocated route tree is cleaned up.

renrichard, 2019-07-23:
free_route_tree(t_rt_node* rt_root) places it on the free_list .
while free_route_tree_timing_structs() completely frees it.

kmurray, 2019-07-30:
Closed with #870

closed 2019-07-30 by kmurray