verilog-to-routing/vtr-verilog-to-routing #847: Travis build cache issue causing sporadic CI failures

kmurray, 2019-07-23:
#### Expected Behaviour

Travis should use a single build step to produce the VTR executables (VPR, ODIN etc.), which are then re-used during the following testing steps (see #619).

#### Current Behaviour

The above works most of the time, but sporadically fails (at least two instances since #619 was merged in January), with the steps subsequent to the build step failing to find any of the generated executables. For example: https://travis-ci.org/verilog-to-routing/vtr-verilog-to-routing/builds/562741339

[image]

where the build succeeded, but the test failed (inspection shows that the required executables were missing).

Since this has sporadically appeared and then gone, its possible this is an issue within the TravisCI infrastructure. But it could potentially be an issue with our configuration (not sure).

#### Possible Solution

One option would be to disable build caching, this would slow down the CI times but side step the issue if its a TravisCI problem.

#### Context

Having inconsistent CI makes it very difficult to fairly evaluate new PRs and detect regressions.

kmurray, 2019-07-23:
@mithro / @acomodi Can you take a look and see if this is a configuration issue on our end?

If its not (ie its a Travis infrastructure issue), we will probably need to disable build caching at least for the interim.

jeanlego, 2019-07-23:
In both cases, I think we should disable it for the time being, if we know that it fixes the issue?

kmurray, 2019-07-24:
I've disabled build artifact caching for now (24e5cb37c) to get the tests running again.

PR #1608 (2021-01-14): ci: add GitHub Actions workflow 'Test', deprecate Travis CI
#### Description

This early work in progress for migrating from Travis CI to GitHub Actions. Not perfect yet, but it seems to work so far. I'm an absolute ignorant on the VTR codebase and on previous CI usage. I just dropped some hours here. Hence, any experienced user please review!

#### Related Issue

- Ref #1493
- Close #847
- Close #1473

#### Motivation and Context

/cc @mithro 

Discussion in verilator: https://github.com/verilator/verilator/issues/2658

#### How Has This Been Tested?

It's CI plumbing, so self-tested. See bel [... cut]
closed 2021-01-14 by litghost