verilog-to-routing/vtr-verilog-to-routing #848: Odin Verilog support

MohamedElgammal, 2019-07-23:
There are many unsupported constructs in Verilog  that I faced while trying to run a verilog design through Odin. I have created a micro testcase for each of them and they can be found at: https://github.com/MohamedElgammal/Odin-issues

#### Expected Behaviour

#### Current Behaviour

#### Possible Solution

#### Steps to Reproduce

1. 
2. 

#### Context

#### Your Environment

* VTR revision used:
* Operating System and version:
* Compiler version:

jeanlego, 2019-07-23:
Thanks for this!

We are actively refactoring a lot of core components of Odin and we can definitely use more micro-test cases! 
Would you be ok with us including those in our regression suite?

MohamedElgammal, 2019-07-23:
Sure you can.

jeanlego, 2019-07-23:
Thank you! We will raise these as individual issue and reference this issue so you can track the development.

jeanlego, 2019-09-23:
@msohrabi2019 can you verify that the issues are still there? And if so please raise new issue for each missing keyword or feature with a new verilog file to exercise the missing feature. 

When creating the issue please refer to this issue so that it is easier to track.
@MohamedElgammal was kind enough to let us use his verilog files, so you could inspire yourself from these for yours.

Thanks again @MohamedElgammal

jeanlego, 2020-01-30:
@Shreyv could you have a look at the related issues?

Shreyv, 2020-01-30:
@jeanlego sure

jeanlego, 2020-03-09:
@djns99 there are other linked issue here that you can start looking at
@Shreyv can you also start working through them also.
@sdamghan can you do the first review on @djns99 and @Shreyv PR's once they submit them.
Also @sdamghan can you have a look at the benchmark suite? Some new change now display known failing benchmark that should pass as part of Verilog 2005.

Shreyv, 2020-03-09:
@jeanlego Sure I will work on it

sdamghan, 2020-03-09:
@jeanlego Sure, I'll handle it!