verilog-to-routing/vtr-verilog-to-routing #849: SystemVerilog support using Verific

MohamedElgammal, 2019-07-23:
As for know, Odin doesn't support SystemVerilog. I think of using Verific as frontend tool to parse systemVerilog and even verilog for Odin. I tried to do so and parse the designs using Verific then dump a verilog netlist but the problem was that Verific doesn't dump all its data structures in the verilog netlist. However, if you could add a parser to parse verific data structure, you will have a complete verilog/systemverilog support.

mithro, 2019-07-23:
FYI -- You can use a Yosys Commercial License (which is Yosys + Verific library) with VtR to parse SystemVerilog.

MohamedElgammal, 2019-07-23:
Hi Tim,
Yeah I know that flow and already tried it too. But it's main issue is that it maps everything thing to soft logic and never used hard blocks like DSPs and BRAMs.
Regards,
Mohamed.

From: Tim Ansell
Sent: Tuesday, July 23, 5:45 p.m.
Subject: Re: [verilog-to-routing/vtr-verilog-to-routing] SystemVerilog support using Verific (#849)
To: verilog-to-routing/vtr-verilog-to-routing
Cc: Mohamed Elgammal, Author

FYI -- You can use a Yosys Commercial License (which is Yosys + Verific library) with VtR to parse SystemVerilog.

jeanlego, 2019-07-23:
Thanks for this :)

We are actively trying to get up to par with Verilog 2005 and hopefully have a large part of it in soon.
Afterwards, we will definitely have a look as this is a very demanded feature!

mithro, 2019-07-23:
@MohamedElgammal - To map to BRAM + DSPs you need to set up your techmaps in Yosys. See an example for Xilinx Series 7 -- https://github.com/SymbiFlow/symbiflow-arch-defs/tree/master/xc7/techmap

vaughnbetz, 2019-07-25:
@jeanlego verific parses System Verilog, and loads up a netlist. As Mohamed said, he tried dumping that into vanilla structural verilog for Odin-II to parse, but both the yosys+verific dumper and the verific Tcl script dumper leave important info about the design out of their output netlist, so the output isn't complete.

Presumably all the data is indeed in memory in the verific data structures, so Odin-II could link to them, query them, and get the netlist. 

We have an academic license for verific, so we didn't have to pay to get this functionality, and presumably other academic users wouldn't have to either.

So this is a significant project, but is something to consider as an alternative/complementary path to improved language support.

mithro, 2019-07-25:
@vaughnbetz what is the data that you are missing from the Yosys output?

MohamedElgammal, 2019-07-25:
@mithro  rom blocks for example are dumped as empty blocks as shown in this block diagram: 
[image]

mithro, 2019-07-25:
@MohamedEldafrawy What is your yosys script look like?

MohamedEldafrawy, 2019-07-25:
I think you meant to mention @MohamedElgammal 

Sorry for closing the discussion by mistake!

MohamedElgammal, 2019-07-25:
To dump Verilog netlist using yosys passes, I used the following commands:
verific -sv top.sv
verific -import -d top.v <top module>

mithro, 2019-07-25:
Have you tried;
```tcl
verific -sv top.sv
verific -import
write_verilog top.v
```

Might also need a `synth` or similar in there....

MohamedElgammal, 2019-07-28:
@mithro thanks alot. I tried it now and it worked and dumped the memory cells.