verilog-to-routing/vtr-verilog-to-routing #878: If Graphic packages are not installed VTR build fails

acomodi, 2019-07-29:
There is an issue with VTR build as, if the necessary graphics libraries, such as `libcairo2-dev` or `libgtk-3-dev` are not installed in the system, the build failes with the following output error.

```
libs/EXTERNAL/libezgl/include/ezgl/graphics.hpp:27:10: fatal error: cairo.h: No such file or directory
 #include <cairo.h>
          ^~~~~~~~~
compilation terminated.
make[2]: *** [libs/EXTERNAL/libezgl/CMakeFiles/ezgl.dir/src/application.cpp.o] Error 1
make[2]: Leaving directory `$SRC_DIR/build'
make[1]: *** [libs/EXTERNAL/libezgl/CMakeFiles/ezgl.dir/all] Error 2
make[1]: Leaving directory `$SRC_DIR/build'
make: *** [all] Error 2
```

#### Expected Behaviour

If graphics is disabled the build should run successfully

#### Current Behaviour

Build fails if necessary libraries are not available in the system.

#### Possible Solution

Probably the issue is that the `NO_GRAPHICS` macro is not correctly set.

#### Steps to Reproduce

1. Latest version of VTR 31a454988f8b48e375a30361a46110d4a0418b65
2. system that does not have `libcairo2-dev` and `libgtk-3-dev` packages available

#### Context

The issue is that for CI builds which do not require graphics enabled, the build will fail.

#### Your Environment

* VTR revision used: 31a454988f8b48e375a30361a46110d4a0418b65
* Operating System and version: Linux/Ubuntu 18.04
* Compiler version: 7.4.0

kmurray, 2019-07-29:
Thanks for filing this @acomodi 

@tinayang3024 As we discussed with @vaughnbetz, this is probably the next thing to look at.

kmurray, 2019-08-14:
I believe this is now fixed with #911.

@acomodi if you still encounter issues please feel free to re-open.

closed 2019-08-14 by kmurray

acomodi, 2019-08-14:
@kmurray, Confirmed, I have checked it and the CI build without `gtk` does not encounter this issue anymore, thanks.