verilog-to-routing/vtr-verilog-to-routing #888: Unexpected option disp on command line

sarashs, 2019-08-01:
When I add disp on I get the following error:
Error 1: Unexpected option 'disp' on command line

Note that I set the graphic flag as true in the makefile and made sure that all of the libraries are installed. Also make does not produce any errors and creates the following output:

make[1]: Entering directory `/home/arash/VTR-ROOT/libarchfpga'
make[1]: `read_arch' is up to date.
make[1]: Leaving directory `/home/arash/VTR-ROOT/libarchfpga'
g++ -I../libarchfpga/include -Wall -Wpointer-arith -Wcast-qual -D__USE_FIXED_PROTOTYPES__ -ansi -pedantic -Wshadow -Wcast-align -D_POSIX_SOURCE -Wno-write-strings -D EZXML_NOMMAP -D_POSIX_C_SOURCE -O3 OBJ/main.o -o vpr -L. -L/usr/lib/X11 -lm -lvpr -lX11

Note that everything but the display works.

#### Your Environment

* VTR revision used: 7
* Operating System and version: ubuntu 14.04
* Compiler version: gcc/g++ 4.8.4

kmurray, 2019-08-01:
I believe the behaviour of the graphics control options (--disp/--nodisp) changed between VPR 7 and VPR 8.

VPR 7: Graphics opens by default unless the `--nodisp` option is given (there is no `--disp` option)
VPR 8: Graphics is closed by default, unless the `--disp on` option is given (there is no `--nodisp` option)

Hope this helps clarify.

closed 2019-08-01 by kmurray

wujiaxuan, 2020-01-19:
I meet with the same problem.
I install VPR7 with ENABLE_GRAPHICS = true, but the graphics does't open..