verilog-to-routing/vtr-verilog-to-routing #940: Odin II fails if the top module doesn't have any ports

tgorochowik, 2019-08-28:
Odin fails (with various syntax errors) if the top module does not have any ports.

#### Expected Behaviour
Odin should not fail with a syntax error if the syntax is correct.

#### Current Behaviour
Odin fails with various syntax errors for valid verilog files.

#### Possible Solution

#### Steps to Reproduce
Consider the following example:
```
module top();
endmodule
```
This is a perfectly valid verlog syntax. Odin fails with:
```
$ odin_II -V /tmp/min.v -o /tmp/x.blif
--------------------------------------------------------------------
Welcome to ODIN II version 0.1 - the better High level synthesis tools++ targetting FPGAs (mainly VPR)
Email: jamieson.peter@gmail.com and ken@unb.ca for support issues

--------------------------------------------------------------------
High-level synthesis Begin
Parser starting - we'll create an abstract syntax tree. Note this tree can be viewed using Grap Viz (see documentation)
Preprocessing verilog.
error in parsing: (syntax error, unexpected vENDMODULE)::0
```

Another example:
```
module top();
  bit xxx;
endmodule
```

```
$ odin_II -V /tmp/min2.v -o /tmp/x.blif
--------------------------------------------------------------------
Welcome to ODIN II version 0.1 - the better High level synthesis tools++ targetting FPGAs (mainly VPR)
Email: jamieson.peter@gmail.com and ken@unb.ca for support issues

--------------------------------------------------------------------
High-level synthesis Begin
Parser starting - we'll create an abstract syntax tree. Note this tree can be viewed using Grap Viz (see documentation)
Preprocessing verilog.
error in parsing: (syntax error, unexpected ';', expecting '(')::0
```

It seems to always happen if the top module does not have any input or output ports.

jeanlego, 2019-09-04:
Odin will produce an empty blif when no primary IO is given since it cleans up the undriven input nets and output.

While I agree that the error is not explicit enough, and maybe give a warning rather than an error would be more suitable, the result will be an empty blif. 

It would help to get an idea of what you were trying to achieve with the empty top module. Are you expecting an empty blif or something else?

tgorochowik, 2019-09-04:
@jeanlego thank you for your reply.

We are working on an open source System Verilog compliance suite which can be found at https://github.com/SymbiFlow/sv-tests

It's still very early stages, but its goal is to cover the whole of the System Verilog LRM.

The project is currently trying to produce a "status report" for popular open source tools. that support Verilog or SystemVerilog, see: https://symbiflow.github.io/sv-tests/

As you can see we wanted to include Odin in the comparison, the issue is that we try to make the test-cases as minimal as possible (at least for the LRM chapters) and Odin seems to fail most of the tests, but not because of lack of the tested features, but because of other reasons (like the ones from this issue).

My expectation would be that there is no error if the syntax is correct (even if the generated blif is empty). A warning seems reasonable.

PR #1148 (2020-02-15): Odin: fix segfault on empty module

#### Description
fix segfault for modules without ports

#### Related Issue
closes #940

#### Motivation and Context
help Odin pass more symbiflow tests

#### How Has This Been Tested?
odin complete regression

#### Types of changes

- [x] Bug fix (change which fixes an issue)
- [ ] New feature (change which adds functionality)
- [ ] Breaking change (fix or feature that would cause existing functionality to change)

#### Checklist:

- [ ] My change requires a change to the documentation
- [ ] I have updated the documentation accordingly
- [x] I have added te [... cut]
closed 2020-02-15 by jeanlego