verilog-to-routing/vtr-verilog-to-routing #945: Coverity Warnings

vaughnbetz, 2019-09-03:
@litghost : there are two coverity warnings in the latest scan; I think they are in code that was likely to have been recently touched by you.  If not, another possibility is @mustafabbas.

_______________________________________________________________________________________________________
*** CID 203682:  Control flow issues  (DEADCODE)
/home/travis/build/verilog-to-routing/vtr-verilog-to-routing/vpr/src/route/rr_graph.cpp: 481 in build_rr_graph(e_graph_type, const std::vector<t_physical_tile_type, std::allocator<t_physical_tile_type>> &, const DeviceGrid &, t_chan_width, e_switch_block_type, int, std::vector<t_switchblock_inf, std::allocator<t_switchblock_inf>>, int, std::vector<t_segment_inf, std::allocator<t_segment_inf>>&, int, int, int, float, float, e_base_cost_type, bool, bool, const t_direct_inf *, int, int *, int *)()
475             size_t max_dim = std::max(grid.width(), grid.height()) - 2; //-2 for no perim channels
476     
477             seg_details = alloc_and_load_seg_details(&max_chan_width,
478                                                      max_dim, segment_inf,
479                                                      use_full_seg_groups, is_global_graph, directionality,
480                                                      &num_seg_details);
>>>     CID 203682:  Control flow issues  (DEADCODE)
>>>     Execution cannot reach the expression "1" inside this statement: "if ((is_global_graph ? 1 : ...".
481             if ((is_global_graph ? 1 : nodes_per_chan.max) != max_chan_width) {
482                 nodes_per_chan.max = max_chan_width;
483                 *Warnings |= RR_GRAPH_WARN_CHAN_WIDTH_CHANGED;
484             }
485     
486             //TODO: Fix

________________________________________________________________________________________________________
*** CID 203684:  Null pointer dereferences  (FORWARD_NULL)
/home/travis/build/verilog-to-routing/vtr-verilog-to-routing/vpr/src/util/vpr_utils.cpp: 783 in find_most_common_block_type(const DeviceGrid &)()
777             }
778         }
779     
780         if (max_type == nullptr) {
781             VTR_LOG_WARN("Unable to determine most common block type (perhaps the device grid was empty?)\n");
782         }
>>>     CID 203684:  Null pointer dereferences  (FORWARD_NULL)
>>>     Passing null pointer "max_type" to "logical_block_type", which dereferences it.
783         return logical_block_type(max_type);
784     }
785     
786     InstPort parse_inst_port(std::string str) {
787         InstPort inst_port(str);
788     

Full details at:

 https://u2389337.ct.sendgrid.net/wf/click?upn=08onrYu34A-2BWcWUl-2F-2BfV0V05UPxvVjWch-2Bd2MGckcRbAHCEZ8DHqT-2FrK1UDr7mvIT337BBJeoPVU9LcuZ07oPimOMPFsAFFXF3UiO-2BL-2Fm2o-3D_yZ-2FW0SaYYRc8Uzk4ssaRTKD4FVooDWhEFmfzabgRDImmAI5hePf2IWz5QmFleZDN4jlK7vUTEWrV0PwLiWIs6ByrwdjYAJ2mCnXUlecsOnvjyVm7mOFrOfZWqnHlkXK9jl0AnGC-2BhsZLTH3ZY7GHbLp54f0TXxwjyKrrFVbNDgplEgSU0-2FSI57V7g1o-2Boymwnmq7KHEJsEwIuuMZI-2BHFZFeoNbcHoBJGnVGUujruoeI-3D

litghost, 2019-09-04:
@vaughnbetz I just requested read-only access to Coverity Scan, and I'll see if I can figure out what it is looking at.

litghost, 2019-09-04:
```
/home/travis/build/verilog-to-routing/vtr-verilog-to-routing/vpr/src/route/rr_graph.cpp: 481
```
is not code I'm familiar with, so I think @mustafabbas is a better candidate.  If @mustafabbas is also unfamiliar I can dig deeper.  Let me know @vaughnbetz .

vaughnbetz, 2019-09-04:
Thanks Keith.  @renrichard is another possibility since he's been refactoring router code.  Richard, Mustafa: is this near code you've been changing?  Looking for a volunteer to fix the warnings (shouldn't be too hard).

renrichard, 2019-09-04:
I'll volunteer a look at the warnings.

vaughnbetz, 2019-09-05:
Great, thanks Richard.  They may be innocuous, but are best to fix even in that case so we stay coverity clean.

AmirhosseinPoolad, 2025-04-30:
From what I can see, this issue was fixed in #953 and we also don't have any Coverity warnings at the moment. Closing.

closed 2025-04-30 by AmirhosseinPoolad