verilog-to-routing/vtr-verilog-to-routing #970: Add gtk memory leak suppression file for vpr in run_vtr_flow

vaughnbetz, 2019-09-19:
1. gtk has a few known memory leaks. We should have a valgrind suppression file so they don't confuse developers running regression tests. This can best be done in run_vtr_flow.

2. run_vtr_flow calls valgrind on every executable when you set the run with valgrind flag. It shouldn't call it on perl, as that results in spurious leaks detected as well.

#### Expected Behaviour
valgrind does not flag these errors

#### Current Behaviour
valgrind flags invalid errors

#### Possible Solution
Add a suppression file, and have the script pick it up.
Don't call valgrind on perl.

vaughnbetz, 2019-09-19:
@mohamedelgammal:  I was hoping you could fix this one.

MohamedElgammal, 2019-10-16:
Fixed!
I have added a valgrind suppresion file in vtr source tree under vpr directory. I also updated the run_vtr_flow.pl to automatically detects this file.

closed 2019-10-16 by MohamedElgammal

vaughnbetz, 2019-10-16:
Excellent -- thanks!