verilog-to-routing/vtr-verilog-to-routing #971: Clock routing enhancements

vaughnbetz, 2019-09-19:
Things to do to finish off the clock code:

1. Support import of the clock network in an rr_graph.xml. I think this just needs a 
<clock_root node=id> tag to be supported, and to identify the virtual clock sink.

2. Add documentation to the vtr web site.  
  - In the tutorial section, add the example from your thesis with the pictures of the network, showing how to build a rib and spine.
  - In the architecture reference section, add the available tags and their meanings.  Pictures giving small examples would be helpful here too, or at least refer to the tutorial and its pictures.

3. The placer should model the delay of global clock networks as a constant, ideally derived by walking the rr-graph.  But even modeling the delay as 0 ns or a defined constant would be preferable to the current behaviour (that we think is happening, but I have not examined the code) of assuming regular routing (and hence high skew).

4. The timing analyzer currently sends back slack ratios to guide optimization as 
sr = slack/max_arrival_time_for_this_constraint
I believe that is causing a 1.6% critical path degradation from your results, due to the max_arrival_times_increasing with the clock network vs. ideal clocks.
We should try changing this to 
sr = slack / max_datapath_delay_for_this_constraint 
(i.e. remove the clock delay)

For #4, probably need discussion with Kevin about how to do that.

kmurray, 2019-09-20:
For (3) the relevant code is here: https://github.com/verilog-to-routing/vtr-verilog-to-routing/blob/master/vpr/src/place/place.cpp#L2238-L2287 which currently assumes the delay of 'ignored' nets is zero. But presumably when we are routing clocks the nets are no longer ignored and hence would get delays from the standard placement delay model (which at the moment won't account for the global clock network).

mustafabbas, 2020-06-04:
(2) is complete here: PR https://github.com/verilog-to-routing/vtr-verilog-to-routing/pull/1104

vaughnbetz, 2025-05-16:
1 and 2 are done.
3 should definitely be done, and 4 may be useful.  3 shouldn't be very hard.  @amin1377 @saaramahmoudi : would either of you be able to take on 3?

amin1377, 2025-05-16:
@vaughnbetz is most concerned with 3; we can punt on 4 if needed.